MKFirmware/MK4duo #760: Ultratronics board SD only read on reset

danandmary, 2019-11-14:
### Issue description
On Board / LCD SD Card only read on system reset / restart.  SD Card is detected (according to LCD) when inserted and removed, however M20 reports SD Init Fail.  Again, when resetting the board, SD Card is detected and Initialised fine allowing me to print from SD Card.  Have tried commenting SD_DETECT_INVERTED, however no change observed, 

### Firmware version
MK4Duo 4.4.0

### My setup
[Configuration_Pins.h.txt](https://github.com/MKFirmware/MK4duo/files/3844071/Configuration_Pins.h.txt)
[Configuration_Overall.h.txt](https://github.com/MKFirmware/MK4duo/files/3844072/Configuration_Overall.h.txt)

Have you configured MK4duo manually? No, Stock Firmware.

MagoKimbra, 2019-11-14:
Try with decomment #define SD_DISABLED_DETECT.
On LCD you found menu for DISMOUNT and MOUNT SD.

danandmary, 2019-11-14:
OK.  I'll give it a go tomorrow and report back.

danandmary, 2019-11-14:
Related to https://github.com/MKFirmware/MK4duo/issues/702

danandmary, 2019-11-15:
Uncommented #define SD_DISABLED_DETECT did not work, here's the log.  You can see the card is read correctly at the start, I then run M22 to unmount, remove the card, reinsert the card, run M21 and M20,  You can see the error.

09:56:46.967 : Begin file list
09:56:46.967 : Retraction.gcode
09:56:46.967 : eeprom.bin
09:56:46.967 : End file list
09:56:55.119 : N17 M502*18
09:56:55.369 : echo:Factory Settings Loaded
09:56:56.963 : N18 M500*31
09:57:00.120 : echo:Settings Stored (1590 bytes; crc 62754)
09:57:04.449 : N19 M22*43
***********SD Card removed and reinserted********
09:57:17.026 : N20 M21*34
09:57:17.026 : N21 M20*34
09:57:17.151 : Error:SD init fail
09:57:17.151 : SD initialization failed.
09:57:17.151 : Do not reformat the card!
09:57:17.151 : Is the card correctly inserted?
09:57:17.151 : Is chipSelect set to the correct value?
09:57:17.151 : Does another SPI device need to be disabled?
09:57:17.151 : Is there a wiring/soldering problem?
09:57:17.151 : errorCode: 32
09:57:17.151 : Begin file list
09:57:17.167 : End file list

danandmary, 2019-11-15:
Figured it out.  The Full Graphical LCD was interfering with the SD Card.  Disabled the LCD, now cards being read on insert without reset.  Now to figure out how to get both LCD and SD Card on the Ultratronics board.  Not sure it will be possible without the Nextion display as it operates on Serial TX RX rather than SPI.

closed 2019-11-15 by danandmary

butte22, 2019-11-27:
Have you solved the error reading sd card? If you have, can you tell me how?