MKFirmware/MK4duo #761: mk4duo 4.3.9, tmc26xstepper conflicting decleration  error

diegcccco, 2019-11-14:
**Please, follow this template when opening a new issue: this will help us a lot in the process of helping you.**

### Issue description

I'm using tmc2660 stepper driver I couldn't solve a library error please help.
I was using version 4.3.5 of mk4duo had no problems. I switched to mk4duo 4.3.9 and did my configuration, but I get an error about the tmc26xstepper and driver library. I don't know how to fix this. I've done and checked all my configuration settings, but I didn't find the line.

I have dropped my configuration files and library files that gave the error below.

### Compile errors
Severity	Code	Description	Project	File	Line
Error		156:16: error: 'driver' has a previous declaration as 'Driver* driver [13]		C:\Users\Desktop\MK4Duo\src\core\stepper\driver\driver.h	156

Severity	Code	Description	Project	File	Line
Error		156:16: error: 'driver' has a previous declaration as 'Driver* driver [13]		C:\Users\ Desktop\MK4Duo\src\core\stepper\driver\driver.h	156

Severity	Code	Description	Project	File	Line
Error		111:37: error: conflicting declaration 'TMC26XStepper driver [4]		C:\Users\ Desktop\MK4Duo\src\core\stepper\tmc26x\tmc26x.h	111

Severity	Code	Description	Project	File	Line
Error		103:37: error: conflicting declaration 'TMC26XStepper driver [4]		C:\Users\ Desktop\MK4Duo\src\core\stepper\tmc26x\tmc26x.h	103

Severity	Code	Description	Project	File	Line
Error		55:37: error: conflicting declaration 'TMC26XStepper driver [4]		C:\Users\ Desktop\MK4Duo\src\core\stepper\tmc26x\tmc26x.h	55

Severity	Code	Description	Project	File	Line
Error		47:37: error: conflicting declaration 'TMC26XStepper driver [4]		C:\Users\ Desktop\MK4Duo\src\core\stepper\tmc26x\tmc26x.h	47

Severity	Code	Description	Project	File	Line
Error		39:37: error: conflicting declaration 'TMC26XStepper driver [4]		C:\Users\ Desktop\MK4Duo\src\core\stepper\tmc26x\tmc26x.h	39

### Firmware version
MK4duo 4.3.9  and MK4duo 4.4.0 versions that have this issue.

### My setup
[configuration_feature.h.txt](https://github.com/MKFirmware/MK4duo/files/3844907/configuration_feature.h.txt)
[configuration_motor_driver.h.txt](https://github.com/MKFirmware/MK4duo/files/3844908/configuration_motor_driver.h.txt)
[driver.h.txt](https://github.com/MKFirmware/MK4duo/files/3844909/driver.h.txt)
[driver.cpp.txt](https://github.com/MKFirmware/MK4duo/files/3844910/driver.cpp.txt)
[tmc26x.h.txt](https://github.com/MKFirmware/MK4duo/files/3844911/tmc26x.h.txt)

MagoKimbra, 2019-11-14:
Unfortunately with the new driver management as object, the TMC2660 library is not usable. I gave priority to the new TMC drivers and the new library. For now you can not mount 4.3.9 or later until it resolves, if I solve. I'm sorry.

diegcccco, 2019-11-14:
> Ne yazık ki, nesne olarak yeni sürücü yönetimi ile, TMC2660 kütüphanesi kullanılamaz. Yeni TMC sürücülerine ve yeni kütüphaneye öncelik verdim. Şimdilik, çözersem çözülünceye kadar 4.3.9 veya üstünü bağlayamazsınız. Üzgünüm.

Is it impossible to control the tmc2660 stepper motor control with the TMC26XStepper library in mk4duo 4.3.9?
can you tell me what to do to fix it?

diegcccco, 2019-11-14:
> Unfortunately with the new driver management as object, the TMC2660 library is not usable. I gave priority to the new TMC drivers and the new library. For now you can not mount 4.3.9 or later until it resolves, if I solve. I'm sorry.

Is it impossible to control the tmc2660 stepper motor control with the TMC26XStepper library in mk4duo 4.3.9?
can you tell me what to do to fix it?

carlosserd, 2019-11-20:
I'm having the same problem. I couldn't configure it with some kind of tmc26xstepper. Did you fix the problem?

closed 2019-12-05 by diegcccco