MKFirmware/MK4duo #763: tmc26xstepper incompatible with mk4do 4.3.9 and 4.4.0. Can you help in getting errors?

butte22, 2019-11-15:
**Please, follow this template when opening a new issue: this will help us a lot in the process of helping you.**

### Issue description
Add mk4duo 4.4.0 or 4.3.9 a tmc26x stepper drivers? If you have the chance to add an additional can.

Can you please add it? Thank you.

Compile errors
Severity Code Description Project File Line
Error 156:16: error: 'driver' has a previous declaration as 'Driver* driver [13] C:\MK4Duo\src\core\stepper\driver\driver.h 156

Severity Code Description Project File Line
Error 156:16: error: 'driver' has a previous declaration as 'Driver* driver [13] C:\MK4Duo\src\core\stepper\driver\driver.h 156

Severity Code Description Project File Line
Error 111:37: error: conflicting declaration 'TMC26XStepper driver [4] C:\MK4Duo\src\core\stepper\tmc26x\tmc26x.h 111

Severity Code Description Project File Line
Error 103:37: error: conflicting declaration 'TMC26XStepper driver [3] C:\MK4Duo\src\core\stepper\tmc26x\tmc26x.h 103

Severity Code Description Project File Line
Error 55:37: error: conflicting declaration 'TMC26XStepper driver [2] C:\MK4Duo\src\core\stepper\tmc26x\tmc26x.h 55

Severity Code Description Project File Line
Error 47:37: error: conflicting declaration 'TMC26XStepper driver [1] C:\MK4Duo\src\core\stepper\tmc26x\tmc26x.h 47

Severity Code Description Project File Line
Error 39:37: error: conflicting declaration 'TMC26XStepper driver [0] C:\MK4Duo\src\core\stepper\tmc26x\tmc26x.h 39

Firmware version
MK4duo 4.3.9 and MK4duo 4.4.0 versions that have this issue.

My setup

danandmary, 2019-11-16:
Hasnt this already been reported, about 3 times?   For the moment, change your stepper driver definitions to DRV8825 (even though you're using TMC drivers).  You won't get the finite control of your TMC steppers, however interpolated step smoothing will still occur.  Once the problem is fixed, change back.

butte22, 2019-11-16:
> Hasnt this already been reported, about 3 times? For the moment, change your stepper driver definitions to DRV8825 (even though you're using TMC drivers). You won't get the finite control of your [...]

Thank you but i have no possibility to change. If you have a chance to help, can you help?

carlosserd, 2019-11-20:
I'm having the same problem. I couldn't configure it with some kind of tmc26xstepper. Did you fix the problem?

butte22, 2019-11-20:
No, unfortunately I could not wait mk4duo. I tried DRV8825 but it didn't.
I am sorry . I tried all the ways, but I couldn't.

danandmary, 2019-11-21:
When you changed your stepper driver definitions, and compiled, what was the error? Can you copy and paste the log?  DRV8825 definitions (with TMC drivers) work fine for me.

butte22, 2019-11-27:
I was able to install TMC2660 and had a test problem.
Thank you for your help :)

MagoKimbra, 2019-12-02:
The TMC2660 no longer supports them too old. I'm sorry.

closed 2019-12-02 by MagoKimbra