MKFirmware/MK4duo #764: Ultratronics Pro Can't read SD while LCD enabled

danandmary, 2019-11-21:
### Issue description
SD Card reader (on board or via LCD) works fine providing Reprapworld Graphical LCD (or Reprapdiscout full graphic LCD) is NOT enabled.  As soon as I enable the graphical LCD, I can use the LCD, however I cannot read or initialise the SD card either on board or via the LCD.

Here is the log file when I attempt to print a file on an SD card 
15:56:43.117 : echo:enqueueing "M23 hand_ok.gcode"
15:56:43.117 : echo:enqueueing "M24"
15:56:43.133 : File opened:hand_ok.gcode Size:19185596
15:56:43.133 : //action:resume
15:56:43.133 : //action:prompt_end
15:56:43.133 : //action:prompt_begin Resuming SD
15:56:43.133 : //action:prompt_button Dismiss
15:56:43.133 : //action:prompt_show
15:56:43.133 : Error:SD read error

### Firmware version
4.4.0

### My setup
Ultratronics Pro with Reprapworld Graphical LCD or Reprapdiscount Full Graphic LCD

Have you configured MK4duo manually? Then write here a list of the settings that may be related to the issue:

Relevant Config
#define SDSUPPORT
//#define USB_FLASH_DRIVE_SUPPORT 
#define ADVANCED_SD_COMMAND
//#define SD_HALF_SPEED
//#define SD_QUARTER_SPEED
//#define SD_EIGHTH_SPEED
#define SD_SIXTEENTH_SPEED // have tried all speeds, same result
//#define SD_CHECK_AND_RETRY
//#define SD_EXTENDED_DIR
#define SD_DISABLED_DETECT

Pins :
#if ENABLED(REPRAPWORLD_GRAPHICAL_LCD)

#define SD_DETECT_PIN       60
 // #define LCD_SDSS            59

  #define LCD_PINS_RS         62 //A8
  #define LCD_PINS_ENABLE     75 //MOSI
  #define LCD_PINS_D4         76 //SCK

  #define BTN_EN1             20
  #define BTN_EN2             21
  #define BTN_ENC             64

#endif // REPRAPWORLD_GRAPHICAL_LCD

butte22, 2019-11-27:
Have you solved the error reading sd card? If you have, can you tell me how?

diegcccco, 2019-12-18:
Hello @danandmary ,   . I'm having the same problem. Can you help me?

HansPrint, 2020-10-29:
Hi,
Having similar issue. Using repetier development sw 1.0.5d on Ultratronics pro 1.0 + LCD 12864 which has SD slot.
Any news on this issue ?
Regards,
Hans