MKFirmware/MK4duo #766: Rumba32v4.4 compilation error

lei111, 2019-11-24:
from C:\Users\MYHOME\Desktop\MK4duo\MK4duo.ino:352:
sketch\src/core/stepper/driver/driver.h: At global scope:
sketch\src/core/stepper/driver/driver.h:172:11: error: member 'Driver driver_t::<unnamed union>::<unnamed struct>::x2' with constructor not allowed in anonymous aggregate
         , x2
           ^~
Multiple libraries were found for "Wire.h"
 Used: C:\Users\MYHOME\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.6.0\libraries\Wire
Multiple libraries were found for "SPI.h"
 Used: C:\Users\MYHOME\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.6.0\libraries\SPI
Multiple libraries were found for "Adafruit_NeoPixel.h"
 Used: C:\Users\MYHOME\Documents\Arduino\libraries\Adafruit_NeoPixel

Hope to get help, thank you very much.

MagoKimbra, 2019-11-24:
Sorry for now not use X2.

lei111, 2019-11-24:
Multiple libraries were found for "Wire.h"
Used: C:\Users\MYHOME\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.6.0\libraries\Wire
Multiple libraries were found for "SPI.h"
Excuse me, what does this mean and how to solve it?

MagoKimbra, 2019-11-24:
First, disable the warnings of arduino ide.
The errors of double libraries by themselves if it has other errors, otherwise it does not give them.
Try the new fixes I put in to see if it works.

lei111, 2019-11-25:
Very sad, still not successful
Is there a known good platformio.ini env setup for Marlin that compiles properly with RUMBA32?

I think this would be a good addition to the WiKi documentation, as platformIO tends to be more straightforward to use once set up properly, and more portable as it does not depend on an existing properly configured Arduino IDE.

MagoKimbra, 2019-11-25:
I have the Rumba32 with tmc2130 driver and everything works ... So it must be something special like X2 that bores me.
Please attach your configuration I try.

lei111, 2019-11-25:
I just didn't set X2, but it still appears, corresponding to multiple libraries spi.h.
All configurations are default, only the motherboard model is rumba32
thank you very much for your help

MagoKimbra, 2019-11-25:
The multi-library error is a known problem in the STM32 library, but if it has no other errors it passes it, quiet. The other errors are the problem. If they are just warning, you need to disable the IDE warnings.
IDE has a bug that if the warnings are on none, you have to put another value, close IDE, reopen IDE and put no one.
If instead there are other errors, if you write me errors check or send me the configuration so I check myself.

lei111, 2019-11-25:
In file included from C:\Users\MYHOME\AppData\Local\Temp\arduino_build_587974\sketch\src/platform/HAL_STM32/HAL.h:43,
                 from C:\Users\MYHOME\AppData\Local\Temp\arduino_build_587974\sketch\src/platform/platform.h:57,
                 from C:\Users\MYHOME\AppData\Local\Temp\arduino_build_587974\sketch\MK4duo.h:117,
                 from C:\Users\MYHOME\Desktop\MK4duo\MK4duo.ino:358:
C:\Users\MYHOME\AppData\Local\Temp\arduino_build_587974\sketch\src/platform/HAL_STM32/HAL_timers.h:149:8: error: 'HardwareTimer' does not name a type; did you mean 'HardwareSerial'?
 extern HardwareTimer *MK_step_timer;
        ^~~~~~~~~~~~~
        HardwareSerial
C:\Users\MYHOME\AppData\Local\Temp\arduino_build_587974\sketch\src/platform/HAL_STM32/HAL_timers.h:154:26: error: variable or field 'Step_Handler' declared void
 extern void Step_Handler(HardwareTimer*);
                          ^~~~~~~~~~~~~
C:\Users\MYHOME\AppData\Local\Temp\arduino_build_587974\sketch\src/platform/HAL_STM32/HAL_timers.h:154:26: error: 'HardwareTimer' was not declared in this scope
C:\Users\MYHOME\AppData\Local\Temp\arduino_build_587974\sketch\src/platform/HAL_STM32/HAL_timers.h:154:26: note: suggested alternative: 'HardwareSerial'
 extern void Step_Handler(HardwareTimer*);
                          ^~~~~~~~~~~~~
                          HardwareSerial
C:\Users\MYHOME\AppData\Local\Temp\arduino_build_587974\sketch\src/platform/HAL_STM32/HAL_timers.h:154:40: error: expected primary-expression before ')' token
 extern void Step_Handler(HardwareTimer*);
                                        ^
C:\Users\MYHOME\AppData\Local\Temp\arduino_build_587974\sketch\src/platform/HAL_STM32/HAL_timers.h: In function 'bool HAL_timer_initialized()':
C:\Users\MYHOME\AppData\Local\Temp\arduino_build_587974\sketch\src/platform/HAL_STM32/HAL_timers.h:167:10: error: 'MK_step_timer' was not declared in this scope
   return MK_step_timer != nullptr;
          ^~~~~~~~~~~~~
C:\Users\MYHOME\AppData\Local\Temp\arduino_build_587974\sketch\src/platform/HAL_STM32/HAL_timers.h: In function 'hal_timer_t HAL_timer_get_current_count(uint8_t)':
C:\Users\MYHOME\AppData\Local\Temp\arduino_build_587974\sketch\src/platform/HAL_STM32/HAL_timers.h:175:36: error: 'MK_step_timer' was not declared in this scope
   return HAL_timer_initialized() ? MK_step_timer->getCount() : 0;
                                    ^~~~~~~~~~~~~
C:\Users\MYHOME\AppData\Local\Temp\arduino_build_587974\sketch\src/platform/HAL_STM32/HAL_timers.h: In function 'void HAL_timer_set_count(uint8_t, hal_timer_t)':
C:\Users\MYHOME\AppData\Local\Temp\arduino_build_587974\sketch\src/platform/HAL_STM32/HAL_timers.h:180:5: error: 'MK_step_timer' was not declared in this scope
     MK_step_timer->setOverflow(count + 1, TICK_FORMAT);
     ^~~~~~~~~~~~~
C:\Users\MYHOME\AppData\Local\Temp\arduino_build_587974\sketch\src/platform/HAL_STM32/HAL_timers.h:180:43: error: 'TICK_FORMAT' was not declared in this scope
     MK_step_timer->setOverflow(count + 1, TICK_FORMAT);

Settings are default

MagoKimbra, 2019-11-25:
But what version of STM32 library do you have? I have the last one ... Here it seems that there are errors just in the name of Harware timers inserted in the last ..

lei111, 2019-11-25:
I am currently installing 1.7.0
I will try to install it on a brand new computer later.

lei111, 2019-11-25:
I just upgraded 1.6 to 1.7
The error has changed.
C:\Users\MYHOME\AppData\Local\Temp\arduino_build_673109\sketch\src\lcd\lcdui.cpp: In static member function 'static void LcdUI::status_printf_P(uint8_t, const char*, ...)':
C:\Users\MYHOME\AppData\Local\Temp\arduino_build_673109\sketch\src\lcd\lcdui.cpp:44:3: error: 'va_start' was not declared in this scope
   va_start(args, message);
   ^~~~~~~~
C:\Users\MYHOME\AppData\Local\Temp\arduino_build_673109\sketch\src\lcd\lcdui.cpp:44:3: note: suggested alternative: 'pwm_start'
   va_start(args, message);
   ^~~~~~~~
   pwm_start
C:\Users\MYHOME\AppData\Local\Temp\arduino_build_673109\sketch\src\lcd\lcdui.cpp:46:3: error: 'va_end' was not declared in this scope
   va_end(args);
   ^~~~~~
C:\Users\MYHOME\AppData\Local\Temp\arduino_build_673109\sketch\src\lcd\lcdui.cpp:46:3: note: suggested alternative: 'rand'
   va_end(args);
   ^~~~~~
   rand
"C:\\Users\\MYHOME\\AppData\\Local\\Arduino15\\packages\\STM32\\tools\\arm-none-eabi-gcc\\8.2.1-1.7/bin/arm-none-eabi-g++" -mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb "@C:\\Users\\MYHOME\\AppData\\Local\\Temp\\arduino_build_673109/sketch/build_opt.h" -c -Os -w -std=gnu++14 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -fno-use-cxa-atexit -MMD "-IC:\\Users\\MYHOME\\Desktop\\MK4duo" "-IC:\\Users\\MYHOME\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.7.0\\cores\\arduino/avr" "-IC:\\Users\\MYHOME\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.7.0\\cores\\arduino/stm32" "-IC:\\Users\\MYHOME\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.7.0\\cores\\arduino/stm32/LL" "-IC:\\Users\\MYHOME\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.7.0\\cores\\arduino/stm32/usb" "-IC:\\Users\\MYHOME\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.7.0\\cores\\arduino/stm32/usb/hid" "-IC:\\Users\\MYHOME\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.7.0\\cores\\arduino/stm32/usb/cdc" "-IC:\\Users\\MYHOME\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.7.0\\system/Drivers/STM32F4xx_HAL_Driver/Inc/" "-IC:\\Users\\MYHOME\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.7.0\\system/Drivers/STM32F4xx_HAL_Driver/Src/" "-IC:\\Users\\MYHOME\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.7.0\\system/STM32F4xx/" "-IC:\\Users\\MYHOME\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.7.0\\system/Middlewares/ST/STM32_USB_Device_Library/Core/Inc" "-IC:\\Users\\MYHOME\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.7.0\\system/Middlewares/ST/STM32_USB_Device_Library/Core/Src" -DSTM32F4xx -DARDUINO=10810 -DARDUINO_RUMBA32_F446VE -DARDUINO_ARCH_STM32 "-DBOARD_NAME=\"RUMBA32_F446VE\"" -DSTM32F446xx -DUSBCON -DUSBD_VID=0x0483 "-DUSB_MANUFACTURER=\"Unknown\"" "-DUSB_PRODUCT=\"RUMBA32_F446VE\"" -DHAL_PCD_MODULE_ENABLED -DUSBD_USE_CDC -DDISABLE_GENERIC_SERIALUSB -DHAL_UART_MODULE_ENABLED "-IC:\\Users\\MYHOME\\AppData\\Local\\Arduino15\\packages\\STM32\\tools\\CMSIS\\5.5.1/CMSIS/Core/Include/" "-IC:\\Users\\MYHOME\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.7.0\\system/Drivers/CMSIS/Device/ST/STM32F4xx/Include/" "-IC:\\Users\\MYHOME\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.7.0\\system/Drivers/CMSIS/Device/ST/STM32F4xx/Source/Templates/gcc/" "-IC:\\Users\\MYHOME\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.7.0\\cores\\arduino" "-IC:\\Users\\MYHOME\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.7.0\\variants\\RUMBA32_F446VE" "-IC:\\Users\\MYHOME\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.7.0\\libraries\\SPI\\src" "-IC:\\Users\\MYHOME\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.7.0\\libraries\\Wire\\src" "C:\\Users\\MYHOME\\AppData\\Local\\Temp\\arduino_build_673109\\sketch\\src\\lcd\\menu\\menu_custom.cpp" -o "C:\\Users\\MYHOME\\AppData\\Local\\Temp\\arduino_build_673109\\sketch\\src\\lcd\\menu\\menu_custom.cpp.o"
Multiple libraries were found for "Wire.h"
 Used: C:\Users\MYHOME\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.7.0\libraries\Wire
Multiple libraries were found for "SPI.h"
 Used: C:\Users\MYHOME\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.7.0\libraries\SPI

MagoKimbra, 2019-11-25:
Please test without lcd, i do not test with normale lcd. I have LCD nextion..