MKFirmware/MK4duo #773: sd detected and sd read error!!

diegcccco, 2019-11-29:
**Please, follow this template when opening a new issue: this will help us a lot in the process of helping you.**

### Issue description

My sd card is plugged in before I run the printer, but it reads when I run it, but when I plug it in again while it is working, it does not read and the files continue to appear in the sd menu.
I select the file from the sd card and initialize it. The sd read error gives error and my nextion screen freezes and does not work when I send data from the usb port. Can you please help? The problem with # 702 has been fixed, but it gives me the same error. What can we help you fix?
### Compile errors
start report:
start
echo:PowerUp
echo:MK4duo_4.3.9
echo:Last Updated: 28-09-2019 | Author: (none, default config)
echo:Compiled: Nov 28 2019
echo:Free Memory: 81884 PlannerBufferBytes: 1280
echo:SD card ok
echo:Factory Settings Loaded
Nextion 4.3" connected!
Testing X connection... OK
Testing Y connection... OK
Testing Z connection... OK
Testing T0 connection... OK

### Firmware version
MK4duo 4.3.9 and mk4duo 4.4.0

### My setup
Nextion 4.3 and ramps fd v1 is using. My english is some bad , I am sorry .
Can you help me?
[Configuration_Feature.txt](https://github.com/MKFirmware/MK4duo/files/3905578/Configuration_Feature.txt)

**Please be patient, we'll be trying to help you as soon as possible.**

MagoKimbra, 2019-12-02:
The card goes dismounted and mounted. If you have the control pin of the inserted SD it does it automatically, but if it is not there or it does not work, better to disable it and do the dismount before removing it and mount it after putting it.
#define SD_DISABLED_DETECT

diegcccco, 2019-12-04:
Thank you for the disabled disabled.Sd did not identify disabled disabled. Let me try this, I'il let you know. I wonder if I have to do anything.

diegcccco, 2019-12-04:
I've defined #SD disable detect but it still gives an SD Read Error error. When I start the 3d printer the eeprom gives a warning that it has changed. Is there a problem with Eeprom? thank you

diegcccco, 2019-12-09:
SD card continues to assume error. I am using Textion 4.3 screen. And I'm using the tmc stepper drivers.
Selecting the gcode file from the sd card makes the first transaction then gives sd read error error. And my nextion screen freezes.
Could the sdcard conflict with Nextion? My English is a bit bad, thank you for helping @MagoKimbra .

diegcccco, 2019-12-09:
Mk4duo 4.3.6 also worked fine. Mk4.3.9 and Mk4duo 4.4.0 give the same error. I can't figure out why. #define SD_HALF_SPEED // OLD VALUE SDSLOW
// # define SD_QUARTER_SPEED
// # define SD_EIGHTH_SPEED
// # define SD_SIXTEENTH_SPEED

I have tried all but none of them. I thought it was related to # 702, I identified and used the sd detect pin. When the SD card is removed, it recognizes when it is inserted, but again gives the error sd read error. I'd be very happy if you could help me. Thank you very much in advance. My English is a bit bad, I'm sorry.

MagoKimbra, 2019-12-09:
No, Nextion use Serial, SD card use SPI port. For testing disabled all TMC in SPI, disconnect al cable from TMC or SPI port... Test SD...

diegcccco, 2019-12-09:
ERRORS:
echo:enqueueing "M24"
File opened:a4_can2rf.gcode Size:25985024
//action:resume
Error: Need Tuning PID
Error: Need Tuning PID
Error: Need Tuning PID
Error: Need Tuning PID
echo:Active Extruder: 0
SD read error
SD read error
SD read error

MagoKimbra, 2019-12-09:
Scusami, ma ce l'hai il lettore SD sulla scheda??

diegcccco, 2019-12-10:
Grazie per aver risposto alla mia domanda @MagoKimbra. // # define TMC_USE_SW_SPI L'ho lasciato inattivo. La cosa che non capisco è che i motori sono in funzione, la scheda SD sta leggendo, posso scegliere gcode. Quando seleziono Gcode, arriva al punto di partenza, quindi fa la prima mossa ma poi aspetta. Dopo aver atteso un po ', fa di nuovo una mossa. E dà costantemente errore di lettura sd.

diegcccco, 2019-12-10:
La scheda SD e il lettore di schede SD non mostrano alcun problema. Mk4duo funziona correttamente nella versione 4.3.6. Ho creato un file gpl da un semplice programma slicer. Potrebbe essere correlato all'affettatrice?

diegcccco, 2019-12-10:
Nella versione 4.4.0, MK4duo si blocca direttamente e non fornisce errori di lettura SD. Nulla appare sul display della porta seriale. Grazie mille per il tuo aiuto

MagoKimbra, 2019-12-10:
Dalla 4.3.8 ho aggiornato la libreria SD, che vede la sd in fat32, ha molte più opzioni. quindi va formattata cosi.

diegcccco, 2019-12-10:
Grazie per aver risposto a @MagoKimbra. Pensi che posso formattare la mia scheda SD? Ne può fare un altro.

diegcccco, 2019-12-12:
Sono inquietante, ma hai trovato qualcosa @ Magokimbra?

diegcccco, 2019-12-12:
Sono inquietante, ma hai trovato qualcosa @MagoKimbra

MagoKimbra, 2019-12-12:
Ma cosa devo trovare scusami?? La libreria fat gestisce tutto lei, se non legge la SD o la porta SPI è bloccata perché hai una periferica SPI e non hai associato il giusto pin CS, oppure il lettore non va, e infine la SD è formattata male. Non c'è altro...

diegcccco, 2019-12-12:
Grazie mille Controllerò di nuovo quello che dici @MagoKimbra

diegcccco, 2019-12-17:
Finora hai risposto alle mie domande @MagoKimbra . Grazie Una domanda sarà più grande.
#define Influisce sulla comunicazione della scheda SD con SERIAL_PORT_2 -2?

MagoKimbra, 2019-12-17:
No, la serial port 2 serve per comunicare con un altro host, tipo il display MKS TFT, oppure se si ha una raspberry collegata in seriale, ma si vuole usare anche il PC in usb.

diegcccco, 2019-12-17:
Grazie mille per il tuo aiuto @MagoKimbra.
I file vengono visualizzati quando accendo la stampante con una scheda SD inserita. Il pin di rilevamento SD viene attivato quando rimuovo e inserisco la scheda SD rimossa. Ma quando inizio, una o due mosse e poi si ferma. Sì, ho eseguito le impostazioni dei pin come nella mia vecchia versione. Ho formattato la mia scheda SD, ma non ha influito di nuovo. Ho fatto tutto quello che hai detto. Sto cercando di risolvere l'errore con la stampante in questo momento. Ho una scheda SD inserita nella scheda sulla porta SPI. Ho installato Cs come un vecchio software.

diegcccco, 2019-12-18:
Il mio italiano è un po 'cattivo @MagoKimbra . Grazie mille per il tuo aiuto. Mi chiedo se tu possa trovare qualcosa. Sto provando a risolvere l'errore adesso

diegcccco, 2019-12-19:
Ciao @MagoKimbra , ti disturbo di nuovo, ma ti chiederò una cosa. Grazie mille per il tuo aiuto. Ho controllato i driver stepper TMC2660 ma non sono riuscito a trovare alcun problema.
// # definisce Ho lasciato inattivo TMC_USE_SW_SPI. Ma invia il comando "-1" quando si rinvia la posizione. Ciò significa che è sbagliato nel senso. Ho fatto quello che hai detto, ma non sono riuscito a risolvere il problema. Dove pensi che dovrei guardare adesso?

diegcccco, 2019-12-19:
Sto preparando il mio file Gcode con semplicità, potrebbe avere qualcosa a che fare con esso?

MagoKimbra, 2019-12-19:
No hai solo un problema di SD non compatibile totalmente o la card difettosa.. Cambia il lettore SD e cambia la card...

diegcccco, 2019-12-20:
Grazie @MagoKimbra  Ho provato oltre 3 diverse schede SD ma non è stato così. Il mio lettore di schede SD funziona nella mia vecchia versione, ma devo sostituirlo?
Sto lavorando con la libreria Tmc2660.h in TMCStepper.h. Driver che funzionano senza problemi.

diegcccco, 2019-12-20:
Grazie mille per il tuo aiuto  @MagoKimbra.
Seguirò ciò che dici e riferirai. C'è un problema nell'uso della libreria in Tmcstepper.h?

diegcccco, 2019-12-23:
Ciao @MagoKimbra . Ti disturbo di nuovo. Questo problema potrebbe essere correlato ad altre impostazioni di configurazione?

diegcccco, 2019-12-23:
Mi puoi lanciare MK4duo 4.3.8 @MagoKimbra?

diegcccco, 2019-12-23:
Sono libero, ma dove posso trovare la versione 4.3.8 di mk4duo @MagoKimbra?

MagoKimbra, 2019-12-23:
On github branch V4.3.8

iosonopersia, 2019-12-23:
@diegcccco 
GitHub branch: https://github.com/MKFirmware/MK4duo/tree/V4_3_8
Online Configurator: https://marlinkimbra.it/configurator/v4_3_8/firmware_configurator.php?lingua=en

diegcccco, 2019-12-23:
Ciao @iosonopersia . Mi dispiace, mi dispiace.
Come posso correggere un errore di runtime?

diegcccco, 2019-12-23:
Grazie @MagoKimbra , @iosonopersia . Mi hai aiutato con tutti i miei problemi finora.

iosonopersia, 2019-12-23:
Non preoccuparti: è solo un'etichetta che usiamo per capire velocemente a quale problema si riferisca ogni issue. Le etichette ci servono per organizzare meglio tutte le issue qui su GitHub, da parte tua non c'è bisogno di fare nulla.

diegcccco, 2019-12-23:
Grazie per l'aiuto @MagoKimbra , @iosonopersia  . Mi dispiace disturbarla tutto il tempo. Quando è possibile risolvere questo problema?