MKFirmware/MK4duo #778: M104 will not honor hotend selection

burner-, 2019-12-07:
After I updated firmware to current github version I got problem that when I enable some of heaters 1-3 it always enable only heater 0 
So command
M104 T1 S220
Will give ok but status messages says after that Recv:  T:77.59 /220 @:255 B:51.88 /50 B@:0 T0:77.59 /220 @0:255 T1:38.59 /0 @1:0 T2:26.76 /0 @2:0 T3:31.78 /0 @3:0 MCU:32.25 max:35.29 F0:0 F1:0  
So it set that temp to T0 instead of T1 
At configuration_basic I have
/***********************************************************************
 ************************** Extruders number ***************************
 ***********************************************************************/
// This defines the number of extruder real or virtual
// 0,1,2,3,4,5,6
#define EXTRUDERS 4

// This defines the number of Driver extruder you have and use
// 0,1,2,3,4,5,6
#define DRIVER_EXTRUDERS 4
/***********************************************************************/

burner-, 2019-12-07:
I use ultratronics board and there is latest configuration what I use: https://github.com/burner-/MK4duo/tree/master/MK4duo

MagoKimbra, 2019-12-07:
The new fw allows the selection of the tools and the dedicated hotend, you must set the tools well with your hotend.
See change log when you download new frimware for news anmd new command...

burner-, 2019-12-07:
You mean M563 and M353 ? Can I hard code those to some configuration? That firmware have tendency to corrupt and lost configuration at flash so all dynamic configuration give only headache and misery. Speaking of that... At Configuration_Temperature there is place for pid settings. But even I use those it always wants pid autotune if it lost configuration. Is there parameter what tells that use those hard coded defaults?

MagoKimbra, 2019-12-07:
My tendency is to arrive at a fully configurable code after compiling. You just have to learn to use some things that offer both the hosts and the firmware itself.
For example M503 returns all the values with relative commands, just make a copy paste in a text file the commands already ready to be given and report the configuration.
With those commands you can for example create a gcode file, recall it with all the settings already ready including pids, make a save in eeprom and you're done.
I use Repetier Host and thanks to the plugin that you find in the fw to copy in the host repetier folder you load the values on the host that allows you to do an export on the PC or import from PC to printer. If you do so in a while you have the printer reconfigured as before.

```
20:37:50.065 : Config:Steps per unit:
20:37:50.065 : Config:  M92 X320.000
20:37:50.066 : Config:  M92 T0 E782.000
20:37:50.066 : Config:Maximum Acceleration (units/s2):
20:37:50.066 : Config:  M201 X1000
20:37:50.066 : Config:  M201 T0 E1000
20:37:50.066 : Config:Maximum feedrates (units/s):
20:37:50.066 : Config:  M203 X300.000
20:37:50.066 : Config:  M203 T0 E40.000
20:37:50.066 : Config:Acceleration (units/s2): P<DEFAULT_ACCELERATION> V<DEFAULT_TRAVEL_ACCELERATION> T* R<DEFAULT_RETRACT_ACCELERATION>
20:37:50.066 : Config:  M204 P500.000 V1000.000
20:37:50.066 : Config:  M204 T0 R5000.000
20:37:50.067 : Config:Advanced: B<DEFAULT_MIN_SEGMENT_TIME> S<DEFAULT_MIN_FEEDRATE> V<DEFAULT_MIN_TRAVEL_FEEDRATE>
20:37:50.067 : Config:  M205 B20000 S0.000 V0.000
20:37:50.067 : Config:Jerk: X<DEFAULT_XJERK> T* E<DEFAULT_EJERK>
20:37:50.067 : Config:  M205 X10.000
20:37:50.067 : Config:  M205 T0 E5.000
20:37:50.067 : Config:Endstop adjustment:
20:37:50.067 : Config:  M666 X-0.12 Y0.00 Z-0.09
20:37:50.067 : Config:Delta Geometry adjustment: ABC<TOWER_*_DIAGROD_ADJ> IJK<TOWER_*_ANGLE_ADJ> UVW<TOWER_*_RADIUS_ADJ>
20:37:50.068 : Config:  M666 A0.000 B0.000 C0.000 I-1.586 J-0.074 K0.000 U0.000 V0.000 W0.000
20:37:50.068 : Config:Delta Geometry adjustment: R<DELTA_RADIUS> D<DELTA_DIAGONAL_ROD>
20:37:50.068 : Config:  M666 R109.90 D217.00
20:37:50.068 : Config:Delta Geometry adjustment: S<DELTA_SEGMENTS_PER_SECOND_PRINT> F<DELTA_SEGMENTS_PER_SECOND_MOVE> L<DELTA_SEGMENTS_PER_LINE>
20:37:50.068 : Config:  M666 S600 F500 L20
20:37:50.068 : Config:Delta Geometry adjustment: O<DELTA_PRINTABLE_RADIUS> P<DELTA_PROBEABLE_RADIUS> H<DELTA_HEIGHT>
20:37:50.069 : Config:  M666 O75.00 P54.00 H209.950
20:37:50.069 : Config:Total number D<driver extruder> E<Extruder> H<Hotend> B<Bed> C<Chamber> <Fan>
20:37:50.069 : Config:  M353 D1 E1 H1 B1 C0 F2
20:37:50.069 : Config:Hotend assignment T<Tool> H<Hotend>
20:37:50.069 : Config:  M563 T0 D0 H0
20:37:50.069 : Config:Heater Sensor parameters: H<Heater> P<Pin> S<Type> A<R25> B<BetaK> C<Steinhart-Hart C> R<Pullup> L<ADC low offset> O<ADC high offset>:
20:37:50.070 : Config:  M305 H0 P36 S4 A100000.0 B4725.0 C0.0000000706 R4700.0 L0 O0
20:37:50.070 : Config:Heater parameters: H<Heater> P<Pin> A<Power Drive Min> B<Power Drive Max> C<Power Max> F<Freq> L<Min Temp> O<Max Temp> U<Use Pid 0-1> I<Hardware Inverted 0-1> R<Thermal Protection 0-1> Q<Pwm Hardware 0-1>:
20:37:50.070 : Config:  M306 H0 P38 A40 B230 C255 F500 L5 O285 U1 I0 Q1 R1
20:37:50.070 : Config:Heater PID parameters: H<Heater> P<Proportional> I<Integral> D<Derivative>:
20:37:50.071 : Config:  M301 H0 P14.46 I1.15 D45.30
20:37:50.071 : Config:Heater Sensor parameters: H<Heater> T<tools> P<Pin> S<Type> A<R25> B<BetaK> C<Steinhart-Hart C> R<Pullup> L<ADC low offset> O<ADC high offset>:
20:37:50.071 : Config:  M305 H-1 T0 P32 S1 A100000.0 B4092.0 C0.0000000000 R4700.0 L0 O0
20:37:50.072 : Config:Heater parameters: H<Heater> T<tools> P<Pin> A<Power Drive Min> B<Power Drive Max> C<Power Max> F<Freq> L<Min Temp> O<Max Temp> U<Use Pid 0-1> I<Hardware Inverted 0-1> R<Thermal Protection 0-1> Q<Pwm Hardware 0-1>:
20:37:50.072 : Config:  M306 H-1 T0 P1 A0 B200 C255 F10 L5 O130 U0 I0 Q1 R1
20:37:50.072 : Config:DHT sensor parameters: P<Pin> S<type 11-21-22>:
20:37:50.072 : Config:  M305 D0 P60 S22
```

For example this is my M503 log. I copy M301 H0 P14.46 I1.15 D45.30 in gcode file or file txt and send this command for setting pid, save into the eeprom with M500 and finish...

burner-, 2019-12-07:
I did not find proper documentation for that changelog
M563: Set Tools heater assignment
T[tools] – Set Tool
D[int] – Set Driver for tool
H[bool] – Set Hotend for tool

But I play around with that and did I guess right that H is not boolean as changelog claims? Its integer number for hotend so commands?:
M563 T0 D0 H0
M563 T1 D1 H1
M563 T2 D2 H2
M563 T3 D3 H3

Will be right syntax?

burner-, 2019-12-07:
Yes I do backups from my printer config and actually these days some of those settings are at my starup script so that it will upload them each time when I connect to printer. For me it feels like workaround for bugs but maybe someone will call it as solution.

MagoKimbra, 2019-12-07:
Ops sorry, H[int] non bool. Ok i fixed it...

burner-, 2020-03-09:
I closed this as resolved. Still I think that it would be good to allow hard code this configuration to firmware but there is so many other issues that better to close this and focust to blocking ones.

closed 2020-03-09 by burner-