MKFirmware/MK4duo #780: SD Card file start issue with Nextion (can select just "Main" and "Refresh")

Iohansan, 2019-12-12:
### Issue description
SD Card is detected and files listed, but not possible to choose and start programm *.gcode
If tap very fast sometimes programms starts or show question about start file.

### Firmware version
MK4duo 4.4.0 and last snapshot 12.12.2019

Arduino DUE.
Nextion 7 enhanced (and 4,3 basic same problem).

carlonb, 2019-12-12:
Are you sure that the filename lenght is no more than 20 chars (extension excluded) ?
I still have this issue, so I do not use long file names.
If you send the same file with Repetier-host no problem about this limitation.

Ratisse, 2019-12-13:
I have same problem, file name  lenght is 11 chars, even when tapping the arrows up and down, Nextion  switches between main and refresh. The only way to select the file is to tap fast several times on it. One thing is more disturbing, after tapping on "Print", and after the heating of the bed, the hotend starts to heat, the nextion displays always the same starting temp (19°) but the hotend never stops heating. Have to rest the printer to show the real temp, very dangerous if you don't have a look at the heating process. The only way I found to avoid this is to pre-heat both bed and hotend, then launch the print

My setup: Nextion 4.3 and MKS Gen 1.4

MagoKimbra, 2019-12-13:
Ok, i check this. For now please return to 4.3.9 version...