MKFirmware/MK4duo #783: Compile error MK4duo 4.4.0 with SPI TMC2130

Erjk2674, 2019-12-22:
### Issue description
Compile error

### Compile errors
C:\Users\eraco\AppData\Local\Temp\arduino_build_175695\sketch\src\core\stepper\tmc\tmc.cpp: In static member function 'static void TMC_Stepper::create_tmc()':

C:\Users\eraco\AppData\Local\Temp\arduino_build_175695\sketch\src\core\stepper\tmc\tmc.cpp:288:140: error: 'TMC_SW_MOSI' was not declared in this scope

       #define TMC_MODEL_DEFINE(ST,A)    do{ if (!driver[ST##_DRV]->tmc)   driver[ST##_DRV]->tmc = new MKTMC(A##_DRV, ST##_CS_PIN, R_SENSE, TMC_SW_MOSI, TMC_SW_MISO, TMC_SW_SCK);    }while(0)

                                                                                                                                            ^

C:\Users\eraco\AppData\Local\Temp\arduino_build_175695\sketch\src\core\stepper\tmc\tmc.cpp:297:7: note: in expansion of macro 'TMC_MODEL_DEFINE'

       TMC_MODEL_DEFINE(X,X);

       ^

C:\Users\eraco\AppData\Local\Temp\arduino_build_175695\sketch\src\core\stepper\tmc\tmc.cpp:288:153: error: 'TMC_SW_MISO' was not declared in this scope

       #define TMC_MODEL_DEFINE(ST,A)    do{ if (!driver[ST##_DRV]->tmc)   driver[ST##_DRV]->tmc = new MKTMC(A##_DRV, ST##_CS_PIN, R_SENSE, TMC_SW_MOSI, TMC_SW_MISO, TMC_SW_SCK);    }while(0)

                                                                                                                                                         ^

C:\Users\eraco\AppData\Local\Temp\arduino_build_175695\sketch\src\core\stepper\tmc\tmc.cpp:297:7: note: in expansion of macro 'TMC_MODEL_DEFINE'

       TMC_MODEL_DEFINE(X,X);

       ^

C:\Users\eraco\AppData\Local\Temp\arduino_build_175695\sketch\src\core\stepper\tmc\tmc.cpp:288:166: error: 'TMC_SW_SCK' was not declared in this scope

       #define TMC_MODEL_DEFINE(ST,A)    do{ if (!driver[ST##_DRV]->tmc)   driver[ST##_DRV]->tmc = new MKTMC(A##_DRV, ST##_CS_PIN, R_SENSE, TMC_SW_MOSI, TMC_SW_MISO, TMC_SW_SCK);    }while(0)

                                                                                                                                                                      ^

C:\Users\eraco\AppData\Local\Temp\arduino_build_175695\sketch\src\core\stepper\tmc\tmc.cpp:297:7: note: in expansion of macro 'TMC_MODEL_DEFINE'

       TMC_MODEL_DEFINE(X,X);

       ^

C:\Users\eraco\AppData\Local\Temp\arduino_build_175695\sketch\src\core\stepper\tmc\tmc.cpp:288:140: error: 'TMC_SW_MOSI' was not declared in this scope

       #define TMC_MODEL_DEFINE(ST,A)    do{ if (!driver[ST##_DRV]->tmc)   driver[ST##_DRV]->tmc = new MKTMC(A##_DRV, ST##_CS_PIN, R_SENSE, TMC_SW_MOSI, TMC_SW_MISO, TMC_SW_SCK);    }while(0)

                                                                                                                                            ^

C:\Users\eraco\AppData\Local\Temp\arduino_build_175695\sketch\src\core\stepper\tmc\tmc.cpp:305:7: note: in expansion of macro 'TMC_MODEL_DEFINE'

       TMC_MODEL_DEFINE(Y,Y);

       ^

C:\Users\eraco\AppData\Local\Temp\arduino_build_175695\sketch\src\core\stepper\tmc\tmc.cpp:288:153: error: 'TMC_SW_MISO' was not declared in this scope

       #define TMC_MODEL_DEFINE(ST,A)    do{ if (!driver[ST##_DRV]->tmc)   driver[ST##_DRV]->tmc = new MKTMC(A##_DRV, ST##_CS_PIN, R_SENSE, TMC_SW_MOSI, TMC_SW_MISO, TMC_SW_SCK);    }while(0)

                                                                                                                                                         ^

C:\Users\eraco\AppData\Local\Temp\arduino_build_175695\sketch\src\core\stepper\tmc\tmc.cpp:305:7: note: in expansion of macro 'TMC_MODEL_DEFINE'

       TMC_MODEL_DEFINE(Y,Y);

       ^

C:\Users\eraco\AppData\Local\Temp\arduino_build_175695\sketch\src\core\stepper\tmc\tmc.cpp:288:166: error: 'TMC_SW_SCK' was not declared in this scope

       #define TMC_MODEL_DEFINE(ST,A)    do{ if (!driver[ST##_DRV]->tmc)   driver[ST##_DRV]->tmc = new MKTMC(A##_DRV, ST##_CS_PIN, R_SENSE, TMC_SW_MOSI, TMC_SW_MISO, TMC_SW_SCK);    }while(0)

                                                                                                                                                                      ^

C:\Users\eraco\AppData\Local\Temp\arduino_build_175695\sketch\src\core\stepper\tmc\tmc.cpp:305:7: note: in expansion of macro 'TMC_MODEL_DEFINE'

       TMC_MODEL_DEFINE(Y,Y);

       ^

C:\Users\eraco\AppData\Local\Temp\arduino_build_175695\sketch\src\core\stepper\tmc\tmc.cpp:288:140: error: 'TMC_SW_MOSI' was not declared in this scope

       #define TMC_MODEL_DEFINE(ST,A)    do{ if (!driver[ST##_DRV]->tmc)   driver[ST##_DRV]->tmc = new MKTMC(A##_DRV, ST##_CS_PIN, R_SENSE, TMC_SW_MOSI, TMC_SW_MISO, TMC_SW_SCK);    }while(0)

                                                                                                                                            ^

C:\Users\eraco\AppData\Local\Temp\arduino_build_175695\sketch\src\core\stepper\tmc\tmc.cpp:313:7: note: in expansion of macro 'TMC_MODEL_DEFINE'

       TMC_MODEL_DEFINE(Z,Z);

       ^

C:\Users\eraco\AppData\Local\Temp\arduino_build_175695\sketch\src\core\stepper\tmc\tmc.cpp:288:153: error: 'TMC_SW_MISO' was not declared in this scope

       #define TMC_MODEL_DEFINE(ST,A)    do{ if (!driver[ST##_DRV]->tmc)   driver[ST##_DRV]->tmc = new MKTMC(A##_DRV, ST##_CS_PIN, R_SENSE, TMC_SW_MOSI, TMC_SW_MISO, TMC_SW_SCK);    }while(0)

                                                                                                                                                         ^

C:\Users\eraco\AppData\Local\Temp\arduino_build_175695\sketch\src\core\stepper\tmc\tmc.cpp:313:7: note: in expansion of macro 'TMC_MODEL_DEFINE'

       TMC_MODEL_DEFINE(Z,Z);

       ^

C:\Users\eraco\AppData\Local\Temp\arduino_build_175695\sketch\src\core\stepper\tmc\tmc.cpp:288:166: error: 'TMC_SW_SCK' was not declared in this scope

       #define TMC_MODEL_DEFINE(ST,A)    do{ if (!driver[ST##_DRV]->tmc)   driver[ST##_DRV]->tmc = new MKTMC(A##_DRV, ST##_CS_PIN, R_SENSE, TMC_SW_MOSI, TMC_SW_MISO, TMC_SW_SCK);    }while(0)

                                                                                                                                                                      ^

C:\Users\eraco\AppData\Local\Temp\arduino_build_175695\sketch\src\core\stepper\tmc\tmc.cpp:313:7: note: in expansion of macro 'TMC_MODEL_DEFINE'

       TMC_MODEL_DEFINE(Z,Z);

       ^
[Configuration_Overall.txt](https://github.com/MKFirmware/MK4duo/files/3992600/Configuration_Overall.txt)
[Configuration_Pins.txt](https://github.com/MKFirmware/MK4duo/files/3992601/Configuration_Pins.txt)

### Firmware version
4.4.0

Manual change
#define TMC_USE_SW_SPI
#define TMC_DEBUG
#define TMC_Z_CALIBRATION

This configuration works with 4.3.9
I'm wrong with some configuration?

MagoKimbra, 2019-12-22:
Ok i fixed.

delete HAS_TRINAMIC &&  in configuration _pin

#if HAS_TRINAMIC && ENABLED(TMC_USE_SW_SPI)
And write only
#if ENABLED(TMC_USE_SW_SPI)

Or download new version.

Erjk2674, 2019-12-22:
Works!!

closed 2019-12-22 by Erjk2674