MKFirmware/MK4duo #784: Unable to build MK4Duo without LCD.

BoHoehne, 2020-01-04:
### Issue description

Unable to build MK4Duo without LCD.

### Firmware version
MK4duo 4.4

### My setup
RUMBA32 with 6x TMC5160 
Delta with sensorless homing and sensorless endstops
NO LCD

Configured with online configuration tool.

The errors are:
===============================================
C:\Users\subausa\AppData\Local\Temp\arduino_build_431257\sketch\src\lcd\lcdui.cpp: In static member function 'static void LcdUI::status_printf_P(uint8_t, const char*, ...)':
C:\Users\subausa\AppData\Local\Temp\arduino_build_431257\sketch\src\lcd\lcdui.cpp:44:3: error: 'va_start' was not declared in this scope
   va_start(args, message);
   ^~~~~~~~
C:\Users\subausa\AppData\Local\Temp\arduino_build_431257\sketch\src\lcd\lcdui.cpp:44:3: note: suggested alternative: 'pwm_start'
   va_start(args, message);
   ^~~~~~~~
   pwm_start
C:\Users\subausa\AppData\Local\Temp\arduino_build_431257\sketch\src\lcd\lcdui.cpp:46:3: error: 'va_end' was not declared in this scope
   va_end(args);
   ^~~~~~
C:\Users\subausa\AppData\Local\Temp\arduino_build_431257\sketch\src\lcd\lcdui.cpp:46:3: note: suggested alternative: 'rand'
   va_end(args);
   ^~~~~~
   rand
"C:\\Users\\subausa\\AppData\\Local\\Arduino15\\packages\\STM32\\tools\\arm-none-eabi-gcc\\8.2.1-1.7/bin/arm-none-eabi-g++" -mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb "@C:\\Users\\subausa\\AppData\\Local\\Temp\\arduino_build_431257/sketch/build_opt.h" -c -Os -std=gnu++14 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -fno-use-cxa-atexit -MMD "-ID:\\3D-Drucker\\Delta\\Rumba32\\MK4duo\\MK4duo" "-IC:\\Users\\subausa\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.7.0\\cores\\arduino/avr" "-IC:\\Users\\subausa\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.7.0\\cores\\arduino/stm32" "-IC:\\Users\\subausa\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.7.0\\cores\\arduino/stm32/LL" "-IC:\\Users\\subausa\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.7.0\\cores\\arduino/stm32/usb" "-IC:\\Users\\subausa\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.7.0\\cores\\arduino/stm32/usb/hid" "-IC:\\Users\\subausa\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.7.0\\cores\\arduino/stm32/usb/cdc" "-IC:\\Users\\subausa\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.7.0\\system/Drivers/STM32F4xx_HAL_Driver/Inc/" "-IC:\\Users\\subausa\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.7.0\\system/Drivers/STM32F4xx_HAL_Driver/Src/" "-IC:\\Users\\subausa\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.7.0\\system/STM32F4xx/" "-IC:\\Users\\subausa\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.7.0\\system/Middlewares/ST/STM32_USB_Device_Library/Core/Inc" "-IC:\\Users\\subausa\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.7.0\\system/Middlewares/ST/STM32_USB_Device_Library/Core/Src" -DSTM32F4xx -DARDUINO=10810 -DARDUINO_RUMBA32_F446VE -DARDUINO_ARCH_STM32 "-DBOARD_NAME=\"RUMBA32_F446VE\"" -DSTM32F446xx -DUSBCON -DUSBD_VID=0x0483 "-DUSB_MANUFACTURER=\"Unknown\"" "-DUSB_PRODUCT=\"RUMBA32_F446VE\"" -DHAL_PCD_MODULE_ENABLED -DUSBD_USE_CDC -DDISABLE_GENERIC_SERIALUSB -DHAL_UART_MODULE_ENABLED "-IC:\\Users\\subausa\\AppData\\Local\\Arduino15\\packages\\STM32\\tools\\CMSIS\\5.5.1/CMSIS/Core/Include/" "-IC:\\Users\\subausa\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.7.0\\system/Drivers/CMSIS/Device/ST/STM32F4xx/Include/" "-IC:\\Users\\subausa\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.7.0\\system/Drivers/CMSIS/Device/ST/STM32F4xx/Source/Templates/gcc/" "-IC:\\Users\\subausa\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.7.0\\cores\\arduino" "-IC:\\Users\\subausa\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.7.0\\variants\\RUMBA32_F446VE" "-IC:\\Users\\subausa\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.7.0\\libraries\\SPI\\src" "-IC:\\Users\\subausa\\Documents\\Arduino\\libraries\\TMCStepper\\src" "-IC:\\Users\\subausa\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.7.0\\libraries\\SoftwareSerial\\src" "-IC:\\Users\\subausa\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.7.0\\libraries\\Wire\\src" "C:\\Users\\subausa\\AppData\\Local\\Temp\\arduino_build_431257\\sketch\\src\\lcd\\menu\\menu_delta_calibrate.cpp" -o "C:\\Users\\subausa\\AppData\\Local\\Temp\\arduino_build_431257\\sketch\\src\\lcd\\menu\\menu_delta_calibrate.cpp.o"
Mehrere Bibliotheken wurden für "SPI.h" gefunden
 Benutzt: C:\Users\subausa\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.7.0\libraries\SPI
Mehrere Bibliotheken wurden für "TMCStepper.h" gefunden
 Benutzt: C:\Users\subausa\Documents\Arduino\libraries\TMCStepper
Mehrere Bibliotheken wurden für "SoftwareSerial.h" gefunden
 Benutzt: C:\Users\subausa\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.7.0\libraries\SoftwareSerial
Mehrere Bibliotheken wurden für "Wire.h" gefunden
 Benutzt: C:\Users\subausa\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.7.0\libraries\Wire

===============================================

Any help would be greatly appreciated.

**Please be patient, we'll be trying to help you as soon as possible.**

MagoKimbra, 2020-01-04:
I try with my configuration and not LCD and haven't error..
Please send me your configuration file.

BoHoehne, 2020-01-04:
I hope that helps.

[Configuration_Overall.zip](https://github.com/MKFirmware/MK4duo/files/4022489/Configuration_Overall.zip)