MKFirmware/MK4duo #791: CAN'T BUILD SCARA MECHANISM

annhan, 2020-02-18:
1. Firmware version: v4.4.0
2. Setting: #define MECHANISM MECH_MORGAN_SCARA
3. error: 'float_limit_t' does not name a type

   float_limit_t  base_pos[XYZ]

can you help me check it? thanks

OscarSegura13, 2021-01-12:
I have the same error: error: 'float_limit_t' does not name a type

the error is in the structure of the following library   ....\MK4duo-master\MK4duo-master\MK4duo\src\core\mechanics\scara_mechanics.h

in the line 34:  **float_limits_t** base_pos[XYZ];

I found a very similar structure in ....D\MK4duo-master\MK4duo-master\MK4duo\src\lib\types.h 
in the line 560: typedef struct MinMaxVal<float>     **limit_float_t;** 

They look similar, I thought it was a typo, but they are not the same type when compiling.

can help for this error in a scara robot

ahmedyashar, 2021-09-30:
i have the same problem!