MKFirmware/MK4duo #792: Problem with servo probe

pirat777knj, 2020-02-20:
Hi.
There was a problem on my board. 
on version 4.4.0 the servo probe does not work.
It looks like this: after practicing Home X & Y, the table lowers and nothing more. The probe does not extend.
When issuing commands to the Servo, everything works.
[Configuration_Overall.txt](https://github.com/MKFirmware/MK4duo/files/4232062/Configuration_Overall.txt)
[Configuration_Pins.txt](https://github.com/MKFirmware/MK4duo/files/4232063/Configuration_Pins.txt)

burner-, 2020-03-08:
It looks that with BLtouch there is same problem. With manual commands it works fine. But with G28 it wont debloy probe. If I deploy it first, it will work. But after probing it wont stow it. 
Also after zprobe it will not rise to Z_PROBE_AFTER_PROBING.

burner-, 2020-03-09:
I dont know how to put this words again. This is not first time when I am just amazed what kind programmers are behind of that development. 
https://github.com/MKFirmware/MK4duo/commit/567144515145607fffbbf29073ad6dafb48be7a0

MK4duo/Configuration_Cartesian.h
#define Z_PROBE_SERVO_NR -1
#define Z_SERVO_ANGLES {90,0} // Z Servo Deploy and Stow angles
#define PROBE_SERVO_NR -1
#define PROBE_SERVO_ANGLES {90,0} // Probe Servo Deploy and Stow angles

So some just decide to rename one of parameters without doing any kind sanity checker for it.

MagoKimbra, 2020-03-09:
Sorry but i do not understand???
#define Z_PROBE_SERVO_NR -1 and #define Z_SERVO_ANGLES {90,0} not exist...

burner-, 2020-03-10:
Myt point is that if you rename configuration parameters you should add old and new ones to sanity check script. So that users get proper error messages if they use old parameters. And probably some kind git commit comments are also good when changes break things.  Currently those name changes does not give any error what so ever. With older configuration file Z probes work partially but not properly. Personally I spent 10hour to debug what is wrong. Finally I add enough debug to code to find out that you remove Z_ from begigning of some parameter names. And I also found that it was done in one massive commit without any commit comments. 
Basically you did all poor practices what developer can do with public git project. So pleas learn some best practices and start use comments at commits and those conditional compile scripts are for reason.

MagoKimbra, 2020-03-10:
But this is what the online configurator does, if you configure by hand it is normal that the change of name creates problems for you. But I point to the configurator I can not put controls even for those who do by hand otherwise my work is doubled ...

burner-, 2020-03-10:
Following best practices of development does not double work. It is classical argument at opensource projects. But at long runt it does not add your work to do things right. Proper commits etc. At least keep blacklist of depricated parameters so if you rename something it will give compile error if you or someone else use those parameters. I am probably only one who say this straight to you but not only one who thinks that. When I got that board I got disclaimer with it that sadly it needs marlin fork and that project is mesh. And long rant of these same things of massive commits without comments, breaking more things than fixing, randomly breaking compatibilty at configuration and G code etc.. That was 3 years ago.  I was suspicious that can it really be that bad. But after I have hear that same rant from 4 different people I thoght that maybe I need tell this also to you. So dont shoot messenger but sorry you way to do things is just bad. If you want working community for this project you really need improve that part. 3 easy steps. Do smaller commits, do comments to commits, do sanity chekcs for things what you change. It really does not double your work and probably you will start get more helping hands to that project also.

MagoKimbra, 2020-03-10:
Watch precisely for lack of help from people who do not want or cannot collaborate and from my less and less available time I am dedicating less and less time to this fw. So in a sincere way as you did, I don't waste too much time doing things that are useless to me ... If it's wrong for you, there are other ways to go, walk them without problems ... I can't waste any more time to follow your instructions, as I said the configurator online fixes these name changes by yourself, if you had used the configurator you would not have lost 10h finding the problem ...
I'm sorry but I can't do more than that ...

closed 2020-03-16 by iosonopersia