MKFirmware/MK4duo #794: error: 'class Probe' has no member named 'check_at_point'

burner-, 2020-03-07:
With new git version and BLTOUCH probe I got compile error src\commands\gcode/bedlevel/abl/g29.h:665:66: error: 'class Probe' has no member named 'check_at_point'

burner-, 2020-03-08:
Looks that this compile with AUTO_BED_LEVELING_UBL but not with AUTO_BED_LEVELING_BILINEAR

Djac1974, 2020-03-10:
Hello, I too have compilation faults
I attach my configuration_overhall and configuration_pins 
I use EndstopminZ for homing, with new version it is necessary to uncomment Z_SAFE_HOMING if not it goes up an error to see attachment "1_Z_SAFE_HOMING", but the homing is done with Z probe.
I configured ENABLED_ALL_SIX_ENDSTOP and uncomment #define SD_ABORT_ON_ENDSTOP_HIT with V4.3.9 it's good, but with new version I have this error see the attachment "2_SD_ABORT_ON_ENDSTOP_HIT"
I must delete in configuration pins HAS_TRINAMIC && otherwise I get an error see attachment "3_ # if HAS_TRINAMIC && ENABLED (TMC_USE_SW_SPI)".
Thanks for your help.

[Configuration_overhall_4.4.0.txt](https://github.com/MKFirmware/MK4duo/files/4314734/Configuration_overhall_4.4.0.txt)

[Configuration_Pins_4.4.0.txt](https://github.com/MKFirmware/MK4duo/files/4314736/Configuration_Pins_4.4.0.txt)

[1_Z_SAFE_HOMING.txt](https://github.com/MKFirmware/MK4duo/files/4314708/1_Z_SAFE_HOMING.txt)

[2_SD_ABORT_ON_ENDSTOP_HIT.txt](https://github.com/MKFirmware/MK4duo/files/4314710/2_SD_ABORT_ON_ENDSTOP_HIT.txt)

[3_#if HAS_TRINAMIC && ENABLED(TMC_USE_SW_SPI).txt](https://github.com/MKFirmware/MK4duo/files/4314711/3_.if.HAS_TRINAMIC.ENABLED.TMC_USE_SW_SPI.txt)

MagoKimbra, 2020-03-10:
1 error is sanitychek with probe you must active Z_SAFE_HOMING
3 Error is this
#if ENABLED(TMC_USE_SW_SPI)  //*/4.3.9 supp HAS_TRINAMIC &&
  #define TMC_SW_MOSI        48  //*/51
  #define TMC_SW_MISO        50
  #define TMC_SW_SCK         52
#endif

The // in #define not possibile used... Please delete  //*/4.3.9 supp HAS_TRINAMIC && and   //*/51

Djac1974, 2020-03-12:
Hello, yes I have activated Z_SAFE_HOMING therefore I can no longer use ENDSTOPPULLUP_ZMIN to do the homing, it is done with BLTOUCH in center of heating bed.
// * / is used to comment the lines as it is done in:
#define TEMP_RESIDENCY_TIME 10 // (seconds)
#define TEMP_WINDOW 1 // (degC) Window around target to start the residency timer x degC early.
For point 2: SD_ABORT_ON_ENDSTOP_HIT does not work with version 4.4.0.
The compilation is successful by changing these parameters, operation of machine is different, on which parameter should I intervene to find the same fonctionnality as in version 4.3.9.

MagoKimbra, 2020-03-12:
When you have a probe, it is the probe that is used as a reference for the home, any endstop Z that is left is used only for emergency if the probe does not work. Of course, to make the home of Z it is better to do it in the center or wherever you want by setting X and Y but it must be on the plate, so activate z_safe_homig.
The comments in normal #defins can be there, but not of the definition of the pins, because there are macros that use that #define that give error in compilation ...
What do you mean the behavior is different? Compared to 4.3.9 there is only the possibility to change the number of hotend, bed, hot room, extruders via gcode and save in eeprom, in addition the management of the stm32 processor has been added, but that does not change anything for those who use other processors ...

Djac1974, 2020-03-13:
Thank you magokimbra for your response.
The behavior is different because with version 4.3.9 I used endstopminZ to do homing and it works well, currently homing is done in the center with Bltouch.
There is therefore no solution to use EndstopminiZ to homing with version 4.4.0?

MagoKimbra, 2020-03-13:
But the error was before, with the probe always you have to use the probe to make the zero. Before you used the Z end stop but it was a mistake ... Now I just added a control that warns you that you have not put the z safe homing, but the operation remains the same. The reference must always be the same, if you have the bltouch it is the bltouch that must make the zero for Z and not the end stop otherwise you have a double error between what the end stop brings and the probe brings when you do the ab .

closed 2020-03-16 by iosonopersia