MKFirmware/MK4duo #797: não compila

Briangsouza, 2020-03-29:
Arduino: 1.8.12 (Mac OS X), Placa:"Arduino Mega or Mega 2560, ATmega2560 (Mega 2560)"

sketch/src/feature/bltouch/bltouch.cpp: In static member function 'static void BLTouch::test()':
sketch/src/feature/bltouch/bltouch.cpp:90:37: error: 'PROBE_TEST_PIN' was not declared in this scope
     deploy_state = HAL::digitalRead(PROBE_TEST_PIN);
                                     ^~~~~~~~~~~~~~
sketch/src/feature/bltouch/bltouch.cpp:90:37: note: suggested alternative: 'PROBE_PT_STOW'
     deploy_state = HAL::digitalRead(PROBE_TEST_PIN);
                                     ^~~~~~~~~~~~~~
                                     PROBE_PT_STOW
sketch/src/feature/bltouch/bltouch.cpp:123:42: error: 'PROBE_TEST_PIN' was not declared in this scope
     if (deploy_state != HAL::digitalRead(PROBE_TEST_PIN)) {   // probe triggered
                                          ^~~~~~~~~~~~~~
sketch/src/feature/bltouch/bltouch.cpp:123:42: note: suggested alternative: 'PROBE_PT_STOW'
     if (deploy_state != HAL::digitalRead(PROBE_TEST_PIN)) {   // probe triggered
                                          ^~~~~~~~~~~~~~
                                          PROBE_PT_STOW
exit status 1
Erro compilando para a placa Arduino Mega or Mega 2560

Este relatório teria mais informações com
"Mostrar a saida detalhada durante a compilação"
opção pode ser ativada em "Arquivo -> Preferências"

Briangsouza, 2020-03-29:
#define Z_HOME_DIR  1. ---> error

MagoKimbra, 2020-03-29:
If you have home on max you must have z probe pin for bltouch.

Briangsouza, 2020-03-29:
OK.. #define Z_PROBE_PIN         ORIG_Z_MIN_PIN.  :)

closed 2020-03-29 by iosonopersia