MKFirmware/MK4duo #805: Maybe an error on default setting using old configuration_overall.h

onekk, 2020-04-25:
### Issue description
During compilation, using an old configuration_overall.h
with probe correctly activate (and working in 4.3.6)
An error is raised at compile time.

### Compile errors

In file included from sketch/src/feature/probe/probe.cpp:30:0:
sketch/src/feature/probe/sanitycheck.h:115:4: error: #error "Z_SAFE_HOMING is recommended when homing with a probe. Enable it or comment out this line to continue."
   #error "Z_SAFE_HOMING is recommended when homing with a probe. Enable it or comment out this line to continue."

### Firmware version
4.4.0

Expected behaviour:

Durign the old configuration_overall.h parsing, the proper value has to be set to make the compilation of the new firmware errorless.

Regards

Carlo D.

MagoKimbra, 2020-04-25:
With the probe you must necessarily enable the Z_SAFE_HOMIG, because the home must be done on the plate with the probe, many forgot to activate it thus making the home of Z out of the plate with problems. So I put it on the sanity check to warn that it wasn't enabled.

onekk, 2020-04-26:
Ok but if it is needed when a probe is present, why not activate it automagically?

MagoKimbra, 2020-04-26:
Why do you have to set the X Y parameters of where to make the home of Z, if I automatically enabled the function nobody would know that it makes the setting and then wonders why the home does in that position?

onekk, 2020-04-27:
I don't understand your answer, maybe an answer to someone else post.

iosonopersia, 2020-04-27:
Parlate in italiano che fate prima ed evitate malintesi :-)

onekk, 2020-04-27:
Ciao, 

Ma se è obbligatoria, perché non attivarla subito, magari con un riquadro evidenziato che segnali che si devono mettere le coordinate di un punto sicuro per fare la home?

Sarebbero risparmiate tante richieste di aiuto relative ad "errori di compilazione".

In questi giorni, sto guardando in giro, e effettivamente il tuo è uno dei migliori firmware in giro, per cui quando posso per la stampa 3d lo consiglio caldamente.

Molti si scontrano con le solite cose, che non sto ad elencarti in quanto tu lo sai fin troppo bene.

Il configuratore online è impagabile, ma a volte genera confusione, non so su cosa è basato e in che punti sia migliorabile.

Ti faccio un paio di esempi:

- Se seleziono un driver non programmabile, sarebbe carino che sparissero tutte le impostazione per quei driver.

- Se seleziono una scheda ad 8 bit trovo dei riferimenti a delle opzioni per le 32bit, ovviamente a me non interessano però qualcuno potrebbe erronamente attivarle e crearsi dei problemi

 Tanto per citare solo le prime due cose che mi vengono in mente.

Poi al solito magari sono io che la faccio semplice.

Saluti

Carlo D.

closed 2020-04-27 by onekk