MKFirmware/MK4duo #808: Which serial port shall I use to communicate with the host computer?

astrojhgu, 2020-04-28:
### Issue description
Which Port shall I choose for RUMBA32 when configuring with the [the configure tool](https://marlinkimbra.it/configurator/v4_4_0/firmware_configurator.php)?

By choosing Serial 1, it compiles, but I note that in the scheme file, usart1 is connected to the aux serial pins. 
By choosing the `default port`, a compilation error raises.

### Compile errors
If I choose the default port, a compilation error will raise :
```
sketch/src/core/stepper/driver/driver.h: In member function 'void Driver::printLabel()':
sketch/src/platform/HAL_STM32/hardwareserial/HardwareSerial.h:36:21: error: 'SerialUSB' was not declared in this scope; did you mean 'SerialLP1'?
   36 |   #define MKSERIAL1 SerialUSB
      |                     ^~~~~~~~~
sketch/src/platform/common/communication/communication.h:100:45: note: in expansion of macro 'MKSERIAL1'
  100 |   #define SERIAL_OUT(WHAT,V...)       (void)MKSERIAL1.WHAT(V)
      |                                             ^~~~~~~~~
sketch/src/platform/common/communication/communication.h:107:39: note: in expansion of macro 'SERIAL_OUT'
  107 | #define SERIAL_TXT(txt)               SERIAL_OUT(print, txt)
      |                                       ^~~~~~~~~~
sketch/src/core/stepper/driver/driver.h:85:57: note: in expansion of macro 'SERIAL_TXT'
   85 |     FORCE_INLINE void printLabel()                    { SERIAL_TXT(axis_letter); }
      |                                                         ^~~~~~~~~~
exit status 1
```

If I choose the serial 1, the compilation is OK.

Before posting this issue, however, take a long breath and check this two things:
1. If you're not using the latest version of Arduino IDE, please try again with the latest one.
arduino version 1.8.9
2. Always check if an error (or more than one) starts with `DEPENDENCY ERROR: _something wrong!_`: **DO NOT POST DEPENDENCY ERRORS**, please! They are messages which tell you what config option was set badly. Fix your configuration and try again. If you also get other types of error, please be sure to have fixed dependency errors **before** posting this issue!

Should not be a dependency error

### Firmware version
MK4duo's versions that have this issue.
version 4.4.0
### My setup
If you used the online configurator, copy the content of `Configuration_Overall.h` and `Configuration_Pins.h` in two text files. **Then Drag&Drop them here. DO NOT COPY THE 
[Configuration_Overall.txt](https://github.com/MKFirmware/MK4duo/files/4543469/Configuration_Overall.txt)
[Configuration_Pins.txt](https://github.com/MKFirmware/MK4duo/files/4543470/Configuration_Pins.txt)
(the .h files are renamed to .txt, to meet the requirement of github)
CONTENT DIRECTLY HERE!!!**

Have you configured MK4duo manually? Then write here a list of the settings that may be related to the issue:
No

**Please be patient, we'll be trying to help you as soon as possible.**

MagoKimbra, 2020-04-28:
It gives you error if you have not set well arduino ide with the right configuration of the card.
Usart support must be enabled (generic Serial)
USB support (if enabled) "CDC (generic 'Serial' supersede Usart)

astrojhgu, 2020-04-28:
> It gives you error if you have not set well arduino ide with the right configuration of the card.
> Usart support must be enabled (generic Serial)
> USB support (if enabled) "CDC (generic 'Serial' supersede Usart)

Thank you. That solves my question. Forgive me for the stupid question.

closed 2020-04-28 by iosonopersia