MKFirmware/MK4duo #810: Arduino due board issue

hoodaxd, 2020-05-06:
### Issue description
I am selecting Arduino due (Native USB Port )
and it gives me compile error it want me to select arduino mega
### Compile errors
In file included from sketch\src/conditionals/conditionals_pins.h:26:0,

                 from sketch\MK4duo.h:100,

                 from C:\Users\utd\Downloads\Compressed\MK4duo-master\MK4duo-master\MK4duo\MK4duo.ino:359:

sketch\src/conditionals/../boards/33.h:9:4: error: #error "Oops! Select 'Arduino Mega' in 'Tools > Board.'"

   #error "Oops! Select 'Arduino Mega' in 'Tools > Board.'"

    ^

exit status 1
Error compiling for board Arduino Due (Native USB Port).

### Firmware version
iam using the last version 

### My setup
i used the guide for arduino due 
https://github.com/MKFirmware/MK4duo/blob/master/Documentation/Compilation.md

MagoKimbra, 2020-05-07:
Because the ramps 1.3 need arduino mega, non arduino due...

hoodaxd, 2020-05-07:
i am making 3d printer using arduino due what should i use ?

MagoKimbra, 2020-05-07:
Board for Arduino due...
/**
 * SAM3X8E ARM Cortex M3
 */
#define BOARD_PICCOLO_3D      1400    // PICCOLO_3D ARM 32 Arduino DUE
#define BOARD_RADDS           1401    // RADDS ARM 32 bit board
#define BOARD_RAMPS_FD_V1     1403    // RAMPS-FD V1 ARM 32 bit board
#define BOARD_RAMPS_FD_V2     1404    // RAMPS-FD V2 ARM 32 bit board
#define BOARD_CNCONTROLS_V14  1405    // CNControls V14
#define BOARD_ULTIMAKER4DUE   1407    // Ultimaker Shield + Arduino DUE Aleksandr Varaksa mods (https://iworld4us.com/diy-3d-printer.html)
#define BOARD_RAMPS_SMART_HFB 1411    // RAMPS-SMART (Power outputs: Hotend, Fan, Bed)
#define BOARD_RAMPS_SMART_HHB 1412    // RAMPS-SMART (Power outputs: Hotend0, Hotend1, Bed)
#define BOARD_RAMPS_SMART_HFF 1413    // RAMPS-SMART (Power outputs: Hotend, Fan0, Fan1)
#define BOARD_RAMPS_SMART_HHF 1414    // RAMPS-SMART (Power outputs: Hotend0, Hotend1, Fan)
#define BOARD_RAMPS4DUE       1430    // RAMPS4DUE with AndrewBCN's RAMPS mods (http://forums.reprap.org/read.php?219,479626,page=1) ARM 32 bit board
#define BOARD_RAMPS_17_HFB    1433    // RAMPS 1.7 (Power outputs: Hotend, Fan, Bed)
#define BOARD_ALLIGATOR_R2    1502    // ALLIGATOR R2 ARM 32 bit board
#define BOARD_ALLIGATOR_R3    1503    // ALLIGATOR R3 ARM 32 bit board
#define BOARD_RURAMPS4D_V11   1550    // RuRAMPS4Duo v1.1 (Power outputs: Hotend0, Hotend1, Hotend2, Fan0, Fan1, Bed)
#define BOARD_RURAMPS4D_V13   1551    // RuRAMPS4Duo v1.3 (Power outputs: Hotend0, Hotend1, Hotend2, Fan0, Fan1, Bed)
#define BOARD_ARCHIM2         1590    // UltiMachine Archim2 (with TMC2130 drivers)
#define BOARD_ULTRATRONICS    1705    // Ultratronics v1.0 ARM 32 bit board

hoodaxd, 2020-05-07:
you mean that in my Board.h should be like this :

/**
 * MK4duo Firmware for 3D Printer, Laser and CNC
 *
 * Based on Marlin, Sprinter and grbl
 * Copyright (c) 2011 Camiel Gubbels / Erik van der Zalm
 * Copyright (c) 2020 Alberto Cotronei @MagoKimbra
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE. See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program. If not, see <http://www.gnu.org/licenses/>.
 *
 */
#pragma once

// Macros for board type
#define BOARD_UNKNOWN -1
#define MB(board) (MOTHERBOARD==BOARD_##board)

/**
 * RAMPS 1.3 / 1.4 - ATmega1280, ATmega2560
 */
//#define BOARD_RAMPS_OLD          3    // MEGA/RAMPS up to 1.2
//#define BOARD_RAMPS_13_HFB      33    // RAMPS 1.3 / 1.4 (Power outputs: Hotend, Fan, Bed)
//#define BOARD_RAMPS_13_HHB      34    // RAMPS 1.3 / 1.4 (Power outputs: Hotend0, Hotend1, Bed)
//#define BOARD_RAMPS_13_HFF      35    // RAMPS 1.3 / 1.4 (Power outputs: Hotend, Fan, Fan)
//#define BOARD_RAMPS_13_HHF      36    // RAMPS 1.3 / 1.4 (Power outputs: Hotend0, Hotend1, Fan)
//#define BOARD_RAMPS_13_HHH      37    // RAMPS 1.3 / 1.4 (Power outputs: Hotend0, Hotend1, Hotend2)

/**
 * RAMPS Derivates - ATmega1280, ATmega2560
 */
//#define BOARD_ULTIMAKER          7    // Ultimaker
//#define BOARD_SAINSMART_2IN1    39    // SainSmart 2 in 1
//#define BOARD_MKS_BASE          40    // MKS BASE v1.0
//#define BOARD_MKS_13            47    // MKS GEN v1.3 or 1.4
//#define BOARD_TRIGORILLA        49    // TRIGORILLA (ANYCUBIC)
//#define BOARD_MKS_GEN_L         53    // MKS GEN L
//#define BOARD_AZTEEG_X3         67    // Azteeg X3
//#define BOARD_AZTEEG_X3_PRO     68    // Azteeg X3 Pro
//#define BOARD_ULTIMAKER_OLD     71    // Ultimaker (Older electronics. Pre 1.5.4. This is rare)
//#define BOARD_ULTIMAIN_2        72    // Ultimainboard 2.x (Uses TEMP_SENSOR 20)
//#define BOARD_GT2560_REV_A      74    // Geeetech GT2560 Rev. A
//#define BOARD_GT2560_REV_A_PLUS 75    // Geeetech GT2560 Rev. A+
//#define BOARD_GT2560_V3         76    // Geeetech GT2560 Rev. B for A10(M/D)
//#define BOARD_GT2560_V3_A20     86    // Geeetech GT2560 Rev. B for A20(M/D)
//#define BOARD_3DRAG             77    // 3Drag Controller
//#define BOARD_K8200             78    // Vellemann K8200 Controller (variant of the 3Drag Controller)
//#define BOARD_3DVERTEX          79    // 3DVertex Controller (Velleman K8400)
//#define BOARD_RUMBA             80    // Rumba
//#define BOARD_ANET             100    // Anet board for A2, A6 and A8
//#define BOARD_RAMPS_ENDER_4    243    // Creality Ender-4, CR-8
//#define BOARD_MAKEBOARD_MINI   431    // MakeBoard Mini v2.1.2 is a control board sold by MicroMake

/**
 * Other ATmega1280, ATmega2560
 */
//#define BOARD_CHEAPTRONIC        2    // Cheaptronic v1.0
//#define BOARD_ELEFU_3           21    // Elefu Ra Board (v3)
//#define BOARD_MKS_MINI          41    // MKS MINI 1.0
[... 56 lines cut ...]
#define BOARD_ULTIMAKER4DUE   1407    // Ultimaker Shield + Arduino DUE Aleksandr Varaksa mods (https://iworld4us.com/diy-3d-printer.html)
#define BOARD_RAMPS_SMART_HFB 1411    // RAMPS-SMART (Power outputs: Hotend, Fan, Bed)
#define BOARD_RAMPS_SMART_HHB 1412    // RAMPS-SMART (Power outputs: Hotend0, Hotend1, Bed)
#define BOARD_RAMPS_SMART_HFF 1413    // RAMPS-SMART (Power outputs: Hotend, Fan0, Fan1)
#define BOARD_RAMPS_SMART_HHF 1414    // RAMPS-SMART (Power outputs: Hotend0, Hotend1, Fan)
#define BOARD_RAMPS4DUE       1430    // RAMPS4DUE with AndrewBCN's RAMPS mods (http://forums.reprap.org/read.php?219,479626,page=1) ARM 32 bit board
#define BOARD_RAMPS_17_HFB    1433    // RAMPS 1.7 (Power outputs: Hotend, Fan, Bed)
#define BOARD_ALLIGATOR_R2    1502    // ALLIGATOR R2 ARM 32 bit board
#define BOARD_ALLIGATOR_R3    1503    // ALLIGATOR R3 ARM 32 bit board
#define BOARD_RURAMPS4D_V11   1550    // RuRAMPS4Duo v1.1 (Power outputs: Hotend0, Hotend1, Hotend2, Fan0, Fan1, Bed)
#define BOARD_RURAMPS4D_V13   1551    // RuRAMPS4Duo v1.3 (Power outputs: Hotend0, Hotend1, Hotend2, Fan0, Fan1, Bed)
#define BOARD_ARCHIM2         1590    // UltiMachine Archim2 (with TMC2130 drivers)
#define BOARD_ULTRATRONICS    1705    // Ultratronics v1.0 ARM 32 bit board

/**
 * SAMD21J18
 */
//#define BOARD_MINITRONICS_V2  2706    // Minitronics v2.0

/**
 * STM32 ARM Cortex-M4F
 */
//#define BOARD_RUMBA32_AUS3D   4203    // RUMBA32 STM32F446VET6 based controller from Aus3D
//#define BOARD_RUMBA32_MKS     4204    // RUMBA32 STM32F446VET6 based controller from MKS
//#define BOARD_STEVAL_3DP001V1 4206    // STEVAL-3DP001V1 3D PRINTER BOARD

MagoKimbra, 2020-05-07:
You don't have to touch board.h, the type of board must be done in configuration_basic.h
#define MOTHERBOARD BOARD_RAMPS_13_HFB
Here you have to put the board you are using between those for Arduino due.

hoodaxd, 2020-05-07:
so how myconfiguration_basic.h should be for arduino due

hoodaxd, 2020-05-07:
i changed 
#define MOTHERBOARD BOARD_RAMPS_13_HFB
#define MOTHERBOARD BOARD_PICCOLO_3D
now there is no error ty for help

closed 2020-05-07 by iosonopersia

MagoKimbra, 2020-05-07:
But can you know what card you have ???