MKFirmware/MK4duo #811: Hello,

PhilippeGonze, 2020-05-25:
**Please, follow this template when opening a new issue: this will help us a lot in the process of helping you.**

### Issue description
Tell us here what happened: don't miss any valuable detail, be clear and concise, don't use UPPERCASE or abbreviations which can make understanding what you wrote quite difficult.

### Compile errors
If you have problems compiling MK4duo, report here the error messages Arduino IDE gives you.

Before posting this issue, however, take a long breath and check this two things:
1. If you're not using the latest version of Arduino IDE, please try again with the latest one.
2. Always check if an error (or more than one) starts with `DEPENDENCY ERROR: _something wrong!_`: **DO NOT POST DEPENDENCY ERRORS**, please! They are messages which tell you what config option was set badly. Fix your configuration and try again. If you also get other types of error, please be sure to have fixed dependency errors **before** posting this issue!

### Firmware version
MK4duo's versions that have this issue.

### My setup
If you used the online configurator, copy the content of `Configuration_Overall.h` and `Configuration_Pins.h` in two text files. **Then Drag&Drop them here. DO NOT COPY THE CONTENT DIRECTLY HERE!!!**

Have you configured MK4duo manually? Then write here a list of the settings that may be related to the issue:
- SETTING_CHANGED_1    true
- SETTING_CHANGED_2    25
- SETTING_CHANGED_3    "blablabla"
- ...

_If needed, you'll be asked for other settings you modified._

**Please be patient, we'll be trying to help you as soon as possible.**

PhilippeGonze, 2020-06-03:
@MagoKimbra any ideas ? (X_HOME_DIR not working if set to -1)

MagoKimbra, 2020-06-03:
X_HOME_DIR It works perfectly...

PhilippeGonze, 2020-06-03:
Thanks for your answer, when I use M119, the M119 only answers with the axis with HOME_DIR that are set to 1.
It seems as if the -1 has a "blocking effet".
In fact, if I try X_HOME_DIR and Z_HOME_DIR to 1, the homing works and M119 too (of course in the wrong direction)

MagoKimbra, 2020-06-03:
A ok i see you have minitronics...
in file 2706.h only max pin is defined
#define ORIG_X_MIN_PIN             NoPin
#define ORIG_X_MAX_PIN              8
#define ORIG_Y_MIN_PIN             NoPin
#define ORIG_Y_MAX_PIN              9
#define ORIG_Z_MIN_PIN             NoPin
#define ORIG_Z_MAX_PIN              4

Now i f you use x_min and not x_max you must change pin in this file.
#define ORIG_X_MIN_PIN              8
#define ORIG_X_MAX_PIN            NoPin
#define ORIG_Y_MIN_PIN             NoPin
#define ORIG_Y_MAX_PIN              9
#define ORIG_Z_MIN_PIN             NoPin
#define ORIG_Z_MAX_PIN              4

PhilippeGonze, 2020-06-03:
Awesome ! This worked on the first try :-D 
A very very very big thank you, you can't imagine how long I tried to solve this.
I never could have done it myself.

--> you, Sir, are my hero !

I made a donation on Marlinkimbra.it to thank you
Have a very nice day

PhilippeGonze, 2020-06-03:
OK I might have gotten little bit carried away. 
This is not 100% as it should be.

I set 
#define X_HOME_DIR -1
#define Y_HOME_DIR 1
#define Z_HOME_DIR -1

and as you wrote, I changed 2706.h to 
#define ORIG_X_MIN_PIN 8
#define ORIG_X_MAX_PIN NoPin
#define ORIG_Y_MIN_PIN NoPin
#define ORIG_Y_MAX_PIN 9
#define ORIG_Z_MIN_PIN 4 //because I use a min pin with inductive probe fixed on head
#define ORIG_Z_MAX_PIN NoPin  

Now X and Y home perfectly thanks to you but Z only goes up for a few millimeters and down again resulting in a slight elevation of the head but no change in the Z position returned by repetier.

It shoul normally go down until reaching endstop min as Z_HOME_DIR is -1.

M119 works 100% on all endstops but it does not seem to affect the Z axis behaviour.
I disabled software endstops so it should'nt be it.

Best regards,

MagoKimbra, 2020-06-04:
Please remove the comments from the line.
#define ORIG_Z_MIN_PIN 4 //because I use a min pin with inductive probe fixed on head
Test probe with M119

PhilippeGonze, 2020-06-04:
Ok it is done, I reloaded the firmware but nothing has changed.

G28 homes X and Y correctly, Z goes up for approx 5 mm and down 1-2 mm.
M119 returns the following :
21:17:54.096 : x_min: TRIGGERED
21:17:54.104 : y_max: TRIGGERED
21:17:54.111 : z_min: NOT TRIGGERED
and it reacts correctly if I switch them manually.

Funny stuff, if I home only Z multiple times, it sometimes doesn't go down after going up.
It also sometimes do a little down before going upas usual.

I also tried to restart from sratch on the original file supplied on reprapworld by @brupje
https://github.com/brupje/MK4duo/releases/download/v4.3.7-1/MK4duo-minitronics20.zip
I changed 2706.h as discussed : 
//###ENDSTOP
#define ORIG_X_MIN_PIN 8
#define ORIG_X_MAX_PIN NoPin
#define ORIG_Y_MIN_PIN NoPin
#define ORIG_Y_MAX_PIN 9
#define ORIG_Z_MIN_PIN 4
#define ORIG_Z_MAX_PIN NoPin

And in the firmware on arduino IDE only changing those : 
#define MECHANISM MECH_CARTESIAN
//#define MECHANISM MECH_DELTA
#define ENDSTOPPULLUP_ZMIN    true
#define X_HOME_DIR -1
#define Y_HOME_DIR 1
#define Z_HOME_DIR -1
#define X_MAX_POS 150
#define DEFAULT_Kp {17.33, 40, 40, 40}     // Kp for H0, H1, H2, H3
#define DEFAULT_Ki {3.19, 07, 07, 07}     // Ki for H0, H1, H2, H3
#define DEFAULT_Kd {23.51, 60, 60, 60}     // Kd for H0, H1, H2, H3
#define X_MIN_POS 0
#define Y_MAX_POS 150
#define Y_MIN_POS 0
#define Z_MAX_POS 150
#define DEFAULT_AXIS_STEPS_PER_UNIT   {160, 160, 4056, 160, 625, 625, 625}

Still, M119 returns good answers but the Z axis won't home.

MagoKimbra, 2020-06-04:
Does the probe work with M119? If when it detects triggered it stops it is normal for it to do it 2 mm before reaching the floor depends on its detection distance ...
The problem is that you have not set that you have a probe, but you are using it as a normal endstop, you have to set that you have a fixed type probe and then you adjust everything with the z probe offset.
When you do the home it stops when it detects, but when the G1 Z0,2 command arrives, for example the first layer will drop to that altitude ...

PhilippeGonze, 2020-06-04:
Ok, I changed the following : 

in the firmware : 
#define ENDSTOPPULLUP_ZPROBE  true
#define Z_PROBE_FIX_MOUNTED
#define Z_MIN_PIN           NoPin
#define Z_PROBE_PIN         ORIG_Z_PROBE_PIN

and in 2706.h
#define ORIG_Z_PROBE_PIN           4
#define ORIG_Z_MIN_PIN NoPin

Now, G28 homes X and Y still perfectly and Z does not move any more.
M119 returns the correct state of all stops (Z being identified as a probe)
23:05:28.492 : x_min: NOT TRIGGERED
23:05:28.501 : y_max: NOT TRIGGERED
23:05:28.516 : z_probe: NOT TRIGGERED

 (again I would like to thank you for the awesome support and time on this issue)

MagoKimbra, 2020-06-04:
No, if you put the probe and don't have the classic endstop, the probe becomes the endtop z-min
So you don't have to set the probe pin, that remains Z-min
#define Z_MIN_PIN NoPin
#define Z_PROBE_PIN ORIG_Z_PROBE_PIN
No this is error
#define Z_MIN_PIN ORIG_Z_MIN_PIN 
#define Z_PROBE_PIN No_pin
and in 2706.h
#define ORIG_Z_PROBE_PIN Nopin
#define ORIG_Z_MIN_PIN 4

PhilippeGonze, 2020-06-06:
Allright then, I use the probe as min pin so I changed following your recommendation.
#define Z_MIN_PIN           ORIG_Z_MIN_PIN
#define Z_PROBE_PIN         NoPin
and in 2706.h
#define ORIG_Z_MIN_PIN 4
#define ORIG_Z_PROBE_PIN           NoPin

Now the M119 returns correctly
17:55:34.768 : x_min: NOT TRIGGERED
17:55:34.784 : y_max: NOT TRIGGERED
17:55:34.790 : z_min: NOT TRIGGERED
G28 homes X and Y correctly but Z goes up for a few mm-cm (varying) and then down 1-5mm.

MagoKimbra, 2020-06-06:
Stet for mm Z is it ok?

PhilippeGonze, 2020-06-06:
no, not at all. The extruder ends up a few centimeters above the bed. 
If I G28 Z only, it does exactly the same thing, going up a few cm  then down a few mm resulting in an even higer position but the position indicated by repetier does not change.

I tried to "interrupt" the first movement (up) by triggering the inductive probe but it does not seem to change anything : the printer continues going up and then down then stops.

MagoKimbra, 2020-06-06:
The nozzle goes up to the probe then it goes down if it detects triggered it bumps up and then it goes down slowly until it finds the plane ... The probe you have detects the plane before the nozzle touches the plane depends on its sensitivity at that point you have to adjust the probe offset to ensure that with G1 Z0 it goes on the plane ...

PhilippeGonze, 2020-06-06:
Hmmm I am not sure I made myself clear here.

Let me reorder things by this example : 
1) X and Z are in 0, Y is in 150 both in repetier and physically  (X and Y automatically, Z manually)
2) M119 returns X,Y and Z endstops as triggered
3) I request a G28
4) X and Y home perfectly (X to 0, Y to 150) but...
5) Z does not home, instead it goes up for a few centimeters (no reaction during this movement if I trigger the Z-probe with a metal piece)
6) Z goes down a little bit but not as much as in 5)
7) machine stops and repetier indicates X and Z in 0, Y in 150 wheareas physically, Z is at ex. 35mm.
8) M119 returns X,Y as triggered, Z endstop as not triggered
9) if I ask again G28,X and Y home perfectly, Z goes up and down again, repetier still indicates Z = 0 but physically, it is at ex. 60 mm 

For info, 
#define Z_PROBE_OFFSET_FROM_NOZZLE -1

MagoKimbra, 2020-06-06:
I can't understand ... Manually put Z at 0 ..
give the command G92 Z0 (set the coordinates of Z to 0)
From G1 Z100 and see if it has risen 100 mm

PhilippeGonze, 2020-06-06:
It has risen 100mm exactly :-/ I did a few prints in the meantime by manually setting Z to zero and the dimensions are good so my steps/mm are ok.

MagoKimbra, 2020-06-06:
Ok now a this point give G28 Z0

PhilippeGonze, 2020-06-06:
Ok so I was physically at Z 100, I typed G28 Z0 but in return I get
21:28:19.520 : echo:Home XY first
21:28:19.551 : X:50.00 Y:75.00 Z:100.000 E:0.0000
and nothing moves.
(X and Y are physically at 50 and 75, that is where I do my Z homing)

I went to see if my feature force home XY before Z was active but it is disabled as
 //#define FORCE_HOME_XY_BEFORE_Z

PhilippeGonze, 2020-06-06:
Sorry I forgot to mention that I get in return the 
22:34:19.189 : echo:Home XY first
That I why I checked the FORCE_HOME_XY_BEFORE_Z which is disabled

phongshader, 2020-06-20:
Did this get resolved? I'm having a similar issue but I'm using a BLTouch clone, switching to a real one in a couple of days when it arrives. X and Y home as expected, bed moves away from probe then moves towards probe and stops before the probe is triggered. Triggering the probe manually does not affect the homing of z at all. M119 shows all stops/probes as triggered when they are physically triggered/closed.

PhilippeGonze, 2020-06-30:
@phongshader, thank your for your input. It is nice to see that I am not the only one  :-) 
Sadly the issue is not yet solved.
I contacted @brupje (the designer of the minitronics V2.0 and the firmware on reprapworld) and gave him the link to this issue entry, I am wainting for his answer.
I hope we will be able to solve this together.