MKFirmware/MK4duo #813: corrupted characters on LCD screen

phongshader, 2020-06-19:
### Issue description
characters on LCD screen are corrupted
MKS Rumba32, REPRAP_DISCOUNT_SMART_CONTROLLER
[image]
[image]
[image]

### Compile errors
none. I compiled several times to make sure this was not a one off.

### Firmware version
MK4duo 4.4.0

### My setup
The Beast V2, cartesian, MKS Rumba32, tmc5160, dual extruder, single nozzle, BLTouch clone
[Configuration_Overall.txt](https://github.com/MKFirmware/MK4duo/files/4805445/Configuration_Overall.txt)
[Configuration_Pins.txt](https://github.com/MKFirmware/MK4duo/files/4805447/Configuration_Pins.txt)

phongshader, 2020-06-23:
There are plenty of other issues as well, BLTouch is not functioning as expected, heaters throwing errors and turning off, but I thought I would start with the lcd screen 1st so I can at least see what's going on.

MagoKimbra, 2020-06-23:
You must setting delay for you lcd and board.
//
// Display Full Graphics delay
//
//#define ST7920_DELAY_1  DELAY_NS(0)
//#define ST7920_DELAY_2  DELAY_NS(0)
//#define ST7920_DELAY_3  DELAY_NS(64)

Search online the valor for your display and board.

phongshader, 2020-06-28:
I tried these settings
0,63,125
0,125,0
125,125,125
0,188,0
0,250,0
0,250,250
250,250,250
96,48,600
96,48,715
313,313,313
0,400,0
none fixed the screen, these are values found in Marlin and searches. Any other ideas?

Kremptor, 2020-08-12:
Hi,
I had the same problem with the Rumba32. Change the file 4203.h. in src / bords. At the end of the file you can set the delay values, it should look like this:
// ### IF_BLOCKS
#define ST7920_DELAY_1 DELAY_NS (350)
#define ST7920_DELAY_2 DELAY_NS (350)
#define ST7920_DELAY_3 DELAY_NS (350)

With this change, it worked for me.
I tested six Display RepRapDiscount Full Graphic Smart Controllers with three Rumba32 boards, with two displays it worked without this change, with the others I had graphic problems.

phongshader, 2020-08-13:
still corrupted when I made the changes in Configuration_Overall.h but was not corrupted when I made the changes in 4204.h
Thank you!

closed 2020-08-13 by phongshader