MKFirmware/MK4duo #819: compilation error for chinese delta on mk4duo latest

allegfede, 2020-07-24:
**Please, follow this template when opening a new issue: this will help us a lot in the process of helping you.**

### Issue description
Tell us here what happened: don't miss any valuable detail, be clear and concise, don't use UPPERCASE or abbreviations which can make understanding what you wrote quite difficult.

### Compile errors
In file included from sketch\src\commands\gcode/gcode.h:107:0,

                 from sketch\src\commands\commands.cpp:30:

sketch\src\commands\gcode/delta/g33_type2.h: In function 'float calibration_probe(const xy_pos_t&, bool)':

sketch\src\commands\gcode/delta/g33_type2.h:163:16: error: 'class Probe' has no member named 'check_at_point'

   return probe.check_at_point(xy, stow ? PROBE_PT_STOW : PROBE_PT_RAISE, 0, false);

                ^~~~~~~~~~~~~~

exit status 1

PS: Arduino ide 1.8.9 (last version compatible with my old xp laptop)

### Firmware version
MK4duo 4.4.0

### My setup
printer is a chinese copy of delta kossel xl (marked micromake, i suppose is a XL version of the D1)

MagoKimbra, 2020-07-24:
You haven't select probe.

allegfede, 2020-07-25:
i set "probe smart effector" at the end of "meccaniche section" (i got a microswitch connected to the pivoting effector).
Have i to set another option?

allegfede, 2020-08-07:
now i set "fixed probe" instead of smart effector and i got (with new downloaded firmware):
``In file included from c:\docume~1\v1p3r\impost~1\temp\arduino_build_212340\sketch\mk4duo.h:57:0,

                 from sketch\src\feature\bedlevel\ubl\ubl_motion.cpp:23:

sketch\src\feature\bedlevel\ubl\ubl_motion.cpp: In static member function 'static bool unified_bed_leveling::line_to_destination_segmented(const feedrate_t&)':

sketch\src\feature\bedlevel\ubl\ubl_motion.cpp:329:55: error: 'struct mechanics_data_t' has no member named 'segments_per_second'; did you mean 'segments_per_second_move'?

     uint16_t    segments      = LROUND(mechanics.data.segments_per_second * seconds),       // Preferred number of segments for distance @ feedrate

                                                       ^

c:\docume~1\v1p3r\impost~1\temp\arduino_build_212340\sketch\src/lib/macros.h:303:29: note: in definition of macro 'LROUND'

 #define LROUND(x)   lroundf(x)

                             ^

sketch\src\feature\bedlevel\ubl\ubl_motion.cpp:332:22: error: 'seglimit' was not declared in this scope

     NOMORE(segments, seglimit);                                                             // Limit to minimum segment length (fewer segments)

                      ^~~~~~~~

sketch\src\feature\bedlevel\ubl\ubl_motion.cpp:332:22: note: suggested alternative: 'XYZlimit'

     NOMORE(segments, seglimit);                                                             // Limit to minimum segment length (fewer segments)

                      ^~~~~~~~

                      XYZlimit

exit status 1
Errore durante la compilazione per la scheda Arduino Mega or Mega 2560.