MKFirmware/MK4duo #822: Nextion with arduino atm2560

lkevin271097, 2020-11-28:
Hi everyone!, 
Hi admin pro.
I can't connecting Nextion with atm2560 (Ramps 1,4)
I use uart1(18,19) or uart 2(13,14) but im fail
I think code config fix
Have you help me???

lkevin271097, 2020-11-28:
please help me!!!

onekk, 2020-12-19:
Sadly to say that this firmware is declared dead, by his author.
As you could read in the first page.

The main firmware documentation site is gone.

Nextion is a piece of hardware not much widespread to find help.

From the little I know It was acting as an host and rely on the "main" processor only to get firmware update.

I've not investigated the code, but maybe some old posts on RepRap forum could be of some help.

Regards

Carlo D.